gdsfactory/gdsfactory #2907: max_cellname_length being ignored

joamatab, 2024-06-23:
max_cellname_length being ignored

```python
import gdsfactory as gf

gf.CONF.max_cellname_length = 3 # making the cell name ridiculously short
c = gf.Component('Example_long_name')
print(c.name)
```

tvt173, 2024-06-24:
in principle this is also fixed by #2909 (and related fixes in kfactory for `show`, etc.)

I don't know if it's really practical that anyone actually sets the limit to such a low number, but it's maybe worth noting that
- when `max_cellname_length` == 9, the `@cell` decorator will reduce the name to `_<HASH>`
- when `max_cellname_length` < 9, the `@cell` decorator will basically ignore you
- when `max_cellname_length` < 8, klayout will not shorten cell names to less than 8 characters on save

which is to say, maybe there should be some sort of warning when `max_cellname_length` is set to anything less than 10?
@sebastian-goeldi 

```python
import gdsfactory as gf

gf.CONF.max_cellname_length = 9 # don't make this number smaller than 9...

@gf.cell
def really_loooooooooong_name():
    return gf.Component()
    
c = gf.Component('Example_long_name')
# don't expect this to work... shortening happens on save or @cell creation
# print(c.name)

# cell name will be shortened
c.write_gds("test.gds")

c2 = really_loooooooooong_name()
# this will also work
print(c2.name)
```

sebastian-goeldi, 2024-06-24:
I would suggest that we simply restrict it to a minimum of 9 with

https://docs.pydantic.dev/2.3/api/types/#pydantic.types.PositiveInt

tvt173, 2024-06-25:
sounds great to me

closed 2024-07-07 by joamatab