gdsfactory/gdsfactory #2925: Question (or request): how to best integrate KLayout Python PCells?

klayoutmatthias, 2024-06-29:
Dear all,

I got an user question and like to discuss that with you:

As GDSFactory is based on KLayout now, it should be easy to integrate KLayout-PCells as components. I came up with a code like this:

```
import gdsfactory as gf
import kfactory as kf
import klayout.db as kdb

# install sample PCell "Spiral" from "SpiralLib"
import spiral_pcell

# obtain the core kdb.Layout object
# TODO: should have an API for this
kcl = kf.kcell._get_default_kcl()
ly = kcl.layout

# top-level component
c = gf.Component("cell")

# a GDSFactory component
c2 = gf.Component("child")
c2.add_ref(gf.components.rectangle(size=(1.0, 1.0), layer=(17, 0), centered=True))
c.add_ref(c2).dmove((2.0, 0.0))

# create a PCell instance
params = {
    "l": kdb.LayerInfo(17, 0),
    "w": 0.2,
    "ai": 0,
    "af": 720,
    "ri": 1.0,
    "rf": 3.0
}
ext_cell = ly.create_cell("Spiral", "SpiralLib", params)

# embed as foreign component
c_ext = gf.Component(kdb_cell=ext_cell, name=ext_cell.name)
c.add_ref(c_ext).dmove((-1.0, 0.0))

# write layout
c.write("out.gds")
```

This code an the demo PCell are here: 
[klayout_pcell_test.zip](https://github.com/user-attachments/files/16040246/klayout_pcell_test.zip)

Embedding KLayout Cells appears to be straightforward via the `kdb_cell` argument of `gf.Component`. But 
I wonder if there is a clean API to access the native default `kdb.Layout` object. In the code above I used `_get_default_kcl`
which does not feel right. Once there is a component, I can access `kcl` from the component via the `KCell` interface, but sometimes
it is good to have access to the singleton default layout object.

For completeness I think there should also
be an API to fetch the underlying `kdb.Cell` of a `gf.Component`. I can access `_kdb_cell` from the `KCell` part of the API, but that is a protected attribute and I have not found a public getter.

Finally, if you use GDSFactory from inside the KLayout application, it should even be possible to embed Ruby-based or native PCells from the "Basic" library. 

Thanks,

Matthias

sebastian-goeldi, 2024-06-29:
Hi Matthias,

The native layout object is at `kf.kcl` (the get default one is purely there because python typing doesn't like circular definitions much in conjunction with pydantic). It's also always available as `KCell.kcl.layou`. I kind of tricked a bit, as all the `KCLayout` are actually libraries, but the layout is more or less transparently proxied (i.e. all `__getattribute__` which are not native to gdsfactory/kfactory).

gdsfactory doesn't support it yet, but kfactory can handle multiple layout objects in the form of libraries. klive also happily will load those libraries. Therefore, it should be relatively easy to get in the basic library.

The main pain point is outlined here:

https://github.com/gdsfactory/gdsfactory/issues/2902#issuecomment-2186623122

Tl;dr kfactory deeply relies on pydantic, which is not a python only package but rather uses a rust core library. This would be really hard to remove as many core concepts rely on pydantic models. For Windows (and to some degree Mac) this would be kinda hard to support. I currently do not have a good plan for this. That's one of the reasons I have told various people at FSiC and in the above mentioned issue that it won't be easy to support it fully. At least not in the near term.

Unless you think there is an easy way to ship pydantic (and maybe ruamel.yaml, which is a python only package so worst case can be shipped with the package), I don't see an easy way to support a full GUI flow as of now on all platforms.

klayoutmatthias, 2024-06-29:
Hi @sebastian-goeldi,

thanks for elaborating on this.

What about `_kdb_cell` to get the native `kdb::Cell` from `KCell`? I can access it, but it does not feel right.

I was mainly referring to using GDSFactory standalone. That is a good enough use case already and as long as PCells are written in Python, embedding them should work.

But there is still this project: https://github.com/efabless/sky130_klayout_pdk. It uses GDSFactory inside KLayout. "pure Python klayout inside klayout" is probably not an issue as with KLayout 0.29.x, KLayout natively supports the "klayout" module, so there is a single - and mostly compatible - binary source. I think that this module takes priority over any separate, pure-Python klayout module. I got "efabless/sky130_klayout_pdk" working on GDSFactory with the KLayout backend with @martinjankoehler (https://github.com/martinjankoehler/sky130_klayout_pdk/pulls). This is another reason I am asking here. This device primitive lib seems to be useful.

Matthias

sebastian-goeldi, 2024-06-29:
`_kdb_cell` is the native way to access the pure `Db::Cell`.  It is "hidden" in a private variable because there are things that could interfere or circumvent kfactory internal functions. E.g. if you were to create an instance through there instead of `<<` or `create_inst` (or in gdsfactory case with `add_ref`), the ports of instances will not work anymore (without specifically asking the KCell to rebuild itself).

@mole99 actually asked me whether there is an easy way to integrate it as pure python or ship it with minimal dependencies.

But I don't fully understand what you mean as pure python. There is no way to use a layout object from the python module in the GUI, is there?

Yes in Linux we can just set the `KLAYOUT_PYTHONPATH`, but I think on other platforms that won't work, will it?

Sebastian

mole99, 2024-06-30:
Hi all!

My problem is that KLayout uses the system-wide package installation and many modern distros hinder/warn you from installing packages there.
This is why it is difficult for most users to get the [sky130_klayout_pdk](https://github.com/efabless/sky130_klayout_pdk) PCells working.

My goal would be:

- Convert the sky130_klayout_pdk PCells to KFactory
- Ship KFactory and all its dependencies with sky130_klayout_pdk
- Have the PCells working on any OS without installing any Python dependencies

Unfortunately, as @sebastian-goeldi said, this is not really possible as pydantic is not a pure python only package.

It is great to see, @klayoutmatthias, that you managed to port the PCells to the KLayout-based GDSfactory! Would love to get this in [sky130_klayout_pdk](https://github.com/efabless/sky130_klayout_pdk).

sebastian-goeldi, 2024-06-30:
@klayoutmatthias if I understand your PR on the sky130 correctly, you are loading external gds right?

The best way in this case that I can think of is essentially to build a rpc type of small app into kfactory/gdsfactory. This would be essentially a reverse klive and in there keep a layout object alive while we are creating PCells. You send a request for a cell and remote builds it and saves it.

The only problem I see in this case: If we have a complex cell, the overhead might be big in case of multiple implementations, e.g. if you have 5 hierarchy levels below. Is it possible on save to save only the cell, but keep the (for the saved gds) invalid references? That would be ideal and fastest.

Top notch would be if we could use a shared memory between the python an klayout, but I don't think that's realistic due to version conflicts etc :P

The only other, but quite ugly, solutions would be to compile klayout against mamba. I know that @proppy managed to do it for windows and @flaport for linux, so only MacOS would be truly missing. In these environments it would be easy to install additional packages as we have direct access to the klayout python env.

klayoutmatthias, 2024-06-30:
Hi @sebastian-goeldi,

external GDS was the way the sky130 PDK was built originally. Some "PCells" are actually static cells. Some are GDSFactory-based PCell implementations.

The original code was transferring layouts from GDSFactory space to KLayout space through temporary GDS files. This is super ugly and also the way, these files were loaded was prone to corrupting the internal layout. 

When I tried the PCells with the latest GDSFactory version, I got a number of issues, one of them was that the same Component could not be created twice (name conflict).

My "fix" boils down to this:

* GDSFactory is *only* used for PCell generation, so I clear the internal canvas layout before generating cells. That avoids the name conflicts and provides a clean canvas for individual PCell generation through GDSFactory.
* The transfer of the layouts is done through a copy of the `kdb.Cell` from GDSFactory space to KLayout library space. That is what I needed the internal cell for. No intent to manipulate it :)
* The static leaf cells are also read in a separate layout and transferred the same way. But that code is not GDSFactory-related.

I see the installation issues of GDSFactory however. This will probably limit the potential application of the PDK.

Honestly I think that GDSFactory is too heavy for the purpose of device construction. Devices are usually not very complex things, specifically as the large and complex devices are delivered as static cells anyway. I'd rather look for an alternative way of implementing the device primitives than to try a complex server/client scheme or something similar. 

I tried such an approach with my experimental primitive lib here: https://github.com/klayoutmatthias/sky130A_el/tree/master/python/sky130_primlib.

Taher Kourany also started a port of the PDK without GDSFactory here: https://github.com/TaherKourany/sky130_klayout_pdk. This project stalled, but maybe someone likes to pick it up again.

Best regards,

Matthias

TaherKourany, 2024-07-01:
Hi,

Python Pcell updates -picked up by me- was a showcase on how current PDK
Pcell could be improved in terms of Hierarchy structures
(availability/reuse of base cells, design flexibility) , computation time
reduction, and less memory usage.
Unfortunately, the project stalled as I have not gotten any feedback from
Amro Tarek.

The repo can be found here and waiting for merge approval at Amro's side.
 https://github.com/TaherKourany/sky130_klayout_pdk

Thanks and best regards,
Taher

mole99, 2024-07-01:
Hi @TaherKourany!

Your PCells would fit in well in the Efabless repo. We currently use the PCells from Amro Tork, but since they depend on gdsfactory, they are complicated to set up.
A pure pya-based implementation would be best from an analog designer's point of view.

Can you tell me about the status of your showcase? Are any devices missing compared to the PCells implemented with gdsfactory? What would be needed to bring them up to par?

Do they require any external dependencies? I see `import pandas as pd` in a couple files, but pandas doesn't seem to be actually used, which would be good.

I will have to take a closer look at your work over the next few days!

PS: My email if you'd like to contact me directly, as this is going slightly off topic: leo.moser@efabless.com

Thanks!
Leo

sebastian-goeldi, 2024-07-01:
@klayoutmatthias @mole99 I have (maybe) interesting news. I just tried to package kfactory (as it has leaner requirements and is probably closer to what efabless wants). I found that pydantic is on the msys2 packages, so I downloaded it and essentially made a salt package with all the dependencies + kfactory (for Windows as Linux and Mac have the venv path, and it's harder to package for them (or maybe not and I just don't understand it enough :P). Everything seems to work, except klayout complains about not being able to set an attribute on the Layout object. I am not sure where that comes from. @klayoutmatthias would you be willing to take a quick look?

I am getting the following error:

```
> import kfactory
TypeError: can't apply this __setattr__ to Layout object
  C:\Users\sgoeldi\KLayout\salt\gdsfactory-dependencies\python\kfactory\kcell.py:3641
  C:\Users\sgoeldi\KLayout\salt\gdsfactory-dependencies\python\kfactory\kcell.py:2930
  C:\Users\sgoeldi\KLayout\salt\gdsfactory-dependencies\python\kfactory\kcell.py:5371
  C:\Users\sgoeldi\KLayout\salt\gdsfactory-dependencies\python\kfactory\__init__.py:15
  (eval):1
```

I tried this on a Win11 VM.

Salt Package:
[gdsfactory-dependencies.zip](https://github.com/user-attachments/files/16051902/gdsfactory-dependencies.zip)

sebastian-goeldi, 2024-07-01:
Hmm, interesting. Thanks Matthias!

I'll check it out. There probably is an erroneous setattr that should be done on the `KCLayout` and not directly on the pure `Db::Layout`

klayoutmatthias, 2024-07-01:
I experimented a little with Ubuntu Python versions and could not confirm that the problem is Python version dependent.

When I look into the code of CPython responsible for the error message, I see that it relies on function pointers. I had cases already where function pointers are not the same on Windows, even if they point to the same function. This can happen, if the same function is instantiated multiple times in different DLLs.

So my rough guess is that there is a basic issue with free Python attributes on KLayout objects on Windows :(

Matthias

sebastian-goeldi, 2024-07-01:
No problem! I can work around it. I can set these properties on the KCLayout object. I intended to do this anyways. Thanks for the valuable debugging. I got it to work as far as to where I stripped out too much code :P.

klayoutmatthias, 2024-07-01:
Ah .. I though that was intended :)

Actually it is funny you can set arbitrary attributes on C++ objects wrapped for Python. Those are not persisted, but maybe there is good use for that somewhere.

I think this capability is missing on Windows since a long time or never was there, so maybe no one missed it. I'll keep an eye on that.

Best regards,

Matthias

sebastian-goeldi, 2024-07-02:
Interestingly it works on the standalone module in windows though. I have some Windows users of kfactory and they never complained.

sebastian-goeldi, 2024-07-02:
I can confirm this approach works:

[image]
[image]

so, @mole99 there is still hope ;)

klayoutmatthias, 2024-07-02:
> Interestingly it works on the standalone module in windows though. I have some Windows users of kfactory and they never complained.

This is likely because that is made with MSVC and this would confirm that compiler details play a role here. 

Thanks for that information.

Matthias

mole99, 2024-07-03:
@sebastian-goeldi Cool! I'm watching eagerly :D

lukasc-ubc, 2025-10-04:
I also worked on trying to get GDSfactory to work in KLayout, and found we can do it on Linux and Mac, but not on Windows.  So we gave up.  But if you can figure it out, that would be useful, I think.  Presently we are writing all our PCells in native pya KLayout code.

sebastian-goeldi, 2025-10-04:
It can be done, but essentially needs kicking out any dependency that needs wheel and then packaging them together with the klayout package. I currently don't have the bandwidth to support this, as this would be of negligible for my use cases (or my employer).

klayoutmatthias, 2025-10-05:
Well, Anaconda is basically possible, but I like to keep the standalone application without pulling Anaconda and I don't know if that really solves this issue.

I am personally user of WSL2 - Ubuntu on Windows. The Linux file system is integrated into the Explorer - transferring files between Linux and Windows is simple and Linux shares all resources with Windows. In my experience, the system is stable enough for daily use.

Matthias