gdsfactory/gdsfactory #2928: absorb not working in new gdsfactory?

qt-thomas, 2024-07-01:
**Describe the bug**
I get the following error when calling `c.absorb(ref)`:
`ValueError: The reference you asked to absorb does not exist in this Component.`

**To Reproduce**
What's the code to reproduce the behavior? What commands or code did you write to get the error?
You can add screenshots to help explain your problem.
Make sure you include the code for others to reproduce your issue.
My code is very simple:
```python
comp = gf.read.from_np(
    eps,
    nm_per_pixel=1000 * ls_grid_size,
    threshold=0.5 * (eps_max + eps_min),
)

c = gf.Component("TOP")
ref = c.add_ref(comp)
ref.xmin = wg_length - ls_grid_size / 2  # I understand this will throw a warning
ref.y = 0 # I understand this will throw a warning
c.absorb(ref)
```

**Expected behavior**
No bugs in gf7. All polygons are properly absorbed.

**Suggested fix**
No idea. I tried following where the absorb function went but it is obscured behind some `_kfinst` object.

qt-thomas, 2024-07-01:
Hi @sebastian-goeldi I want a cell/component with no references, just polygons. Would your suggestion achieve this?

sebastian-goeldi, 2024-07-01:
Ah, in that case you can also completely flatten the cell with

`c.flatten()`

The difference is `ref.flatten` absorbs only that reference/instance. `c.flatten()` will flatten everything. Those two are equivalent in the case of there only being one ref, which I think is the case in your example.

Tl;dr if the cell should be flat use `c.flatten()` if only a specific instance should be flattened use `ref.flatten()`

joamatab, 2024-07-01:
nice catch Thomas,

yes, i've been replacing 
```
c.absorb(ref)
```
with 
```
c.flatten()
```

Sebastian, 

ref.flatten() does not seem to work

[image]

sebastian-goeldi, 2024-07-01:
@joamatab 

because `kf.Instance` is overloaded/replaced with ComponentReference. To make it work again, you need to do

`ref._kfinst.flatten()`

I would suggest to overload flatten in ComponentReference to do it.

commit 9543aada (2024-07-01): fix #2928
commit c5f2fe4a (2024-07-01): Merge pull request #2931 from gdsfactory/2928-absorb-not-working-in-new-gdsfactory

fix #2928
closed 2024-07-01 by joamatab