gdsfactory/gdsfactory #2933: klive didn't send data

ninja-lab, 2024-07-01:
I've got KLayout open and some gdsfactory code in python: 
```
c = gf.Component()
c.add_polygon([(-8, -6), (6, 8), (7, 17), (9, 5)], layer=(1, 0))
c.plot()
c.show()
```
I installed the klive plugin in KLayout, but not the gdsfactory one, because the [readme](https://github.com/gdsfactory/klive) said not to. 
From both jupyter-lab and spyder, I get this when running the python code above: 
```
2024-07-01 14:50:46.512 | WARNING  | kfactory.kcell:show:7783 - klive didn't send data, closing
```
The klive indicator in the toolbar of KLayout does start out green, but then goes yellow and stays yellow. 
I've got a blank layout open in KLayout. Anything else I'm missing for this to work?

sebastian-goeldi, 2024-07-01:
I think in this case maybe the gds generated is corrupt or you might be using a very old klayout GUI. Which version of the GUI are you using?

ninja-lab, 2024-07-01:
I'm on KLayout 0.29.1 on Pop!OS 22.04 (basically Ubuntu 22.04). The KLayout page says 0.29.2 is the latest, so mine isn't very old.

sebastian-goeldi, 2024-07-01:
Not sure then. You can try open the macros development in klayout first and then try to run the code. klive should then stop on encountering an error (as long as debug mode is enabled and none of klive's files are in the ignore list

ninja-lab, 2024-07-02:
What code? The same python code?

sebastian-goeldi, 2024-07-02:
yes. But if you open the macros editor in klayout, you will automatically let it run through the debugger and it should properly show errors.

The code should be run in whatever shell or IDE you ran it before

aiyoung, 2025-06-28:
Hello all, 

I am also new to playing around with GDSFactory and am also using Pop!_OS. 

I ran into the same issue as @ninja-lab where Klive wasn't sending data to KLayout. 

The issue seemed to be that I had installed KLayout from the Pop!_OS store rather than snagging the .deb  from https://www.klayout.de/build.html

After uninstalling the version of KLayout from the Pop!_OS store and re-installing KLayout using the .deb, Klive is now able to talk to KLayout. 

Hopefully this comment will help others who find themselves in a similar situation while getting started with GDSFactory.