gdsfactory/gdsfactory #2935: Tutorial error

AhableelTII, 2024-07-02:
Hi, im following the tutorial from the ``gdsfactory`` documentation, specifically the code below

```
import gdsfactory as gf

c = gf.Component()

# Create some new geometry from the functions available in the geometry library
t = gf.components.text("Hello!")
r = gf.components.rectangle(size=(5, 10), layer=(2, 0))

# Add references to the new geometry to c, our blank component
text1 = c.add_ref(t)  # Add the text we created as a reference
# Using the << operator (identical to add_ref()), add the same geometry a second time
text2 = c << t
r = c << r  # Add the rectangle we created

# Now that the geometry has been added to "c", we can move everything around:
text1.dmovey(25)
text2.dmove([5, 30])
text2.drotate(45)
r.dmovex(-15)

print(c)
c.plot()
```
And below is the error i get, is this a bug is there any update? because i belive i have the latest version

```
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
File c:\Users\abdalla.hableel\Desktop\gdsfactory_test.py:16
     [13](file:///C:/Users/abdalla.hableel/Desktop/gdsfactory_test.py:13) r = c << r  # Add the rectangle we created
     [15](file:///C:/Users/abdalla.hableel/Desktop/gdsfactory_test.py:15) # Now that the geometry has been added to "c", we can move everything around:
---> [16](file:///C:/Users/abdalla.hableel/Desktop/gdsfactory_test.py:16) text1.dmovey(25)
     [17](file:///C:/Users/abdalla.hableel/Desktop/gdsfactory_test.py:17) text2.dmove([5, 30])
     [18](file:///C:/Users/abdalla.hableel/Desktop/gdsfactory_test.py:18) text2.drotate(45)

AttributeError: 'ComponentReference' object has no attribute 'dmovey'
```

sebastian-goeldi, 2024-07-02:
Hi,

Are you sure you are on the latest gdsfactory and not version 7? `dmovey` is a new command (due to adapting the naming scheme to klayout).

AhableelTII, 2024-07-02:
> Hi,
> 
> 
> 
> Are you sure you are on the latest gdsfactory and not version 7? `dmovey` is a new command (due to adapting the naming scheme to klayout).
> 
> 
> 
> 

Indeed
I was using the version from vscode store.
Now i got version 8.4 

Thanks

sebastian-goeldi, 2024-07-02:
@joamatab can we update that package in vscode?

joamatab, 2024-07-02:
VS Code extension does not have any python code,

it just adds the pic.yml  watcher button

closed 2024-07-02 by joamatab