gdsfactory/gdsfactory #2943: c.show() not working after new gdsfactory 8 install

Hohniker, 2024-07-04:
Fresh install of Anaconda, KLayout (all 4 packages described in documentation), and gdsfactory 8. 

I run the following code in VScode with an instance of KLayout open to see if KLive works.

import gdsfactory as gf

c = gf.components.straight()

c.show()

I get the following error. Why is it trying to look for git executable? Do I need to set up git just to run the simplest few lines of code?

[image]

Hohniker, 2024-07-04:
Resolved. Put path to git in env variables. But had to restart PC in order for VScode to see the changes. Still strange that it was looking for git.

closed 2024-07-04 by Hohniker

sebastian-goeldi, 2024-07-04:
Just as an fyi, it's looking for the git executable because of git-python. That relies on the git executable because you can use barebone git commands.

sebastian-goeldi, 2024-07-04:
That's indirectly a dependency of kfactory. If it causes many problems we can take it out. But it's nice to setup a proper build folder and set up gds builds there

aroop25, 2024-11-30:
I am learning gdsfactory and I am running into the same problem. I have zero git experience. Can anyone tell me how to put path to git in env variable?

aroop25, 2024-11-30:
> Resolved. Put path to git in env variables. But had to restart PC in order for VScode to see the changes. Still strange that it was looking for git.

It worked for me too!!