gdsfactory/gdsfactory #2950: How to change the layer of the route for the command "gdsfactory.routing.route_bundle_all_angle"？

ZhimingShi, 2024-07-07:
### Hello！I have two questions, the first one is，how do I change the connected layers in the 'gdsfactory.routing.route_bundle_all_angle' command in gdsfactory v8?
[image]
It doesn't have a 'cross_section' parameter and I don't know how to modify it .

### The second problem is that there seems to be an error in the path length returned by the gf.routing.route_single command, the code to reproduce the error is as follows:
import gdsfactory as gf
import numpy as np
c = gf.Component()
waveguide_width = 1
ring_width = 1
ring_radius = 114
layer = (2,0)
couple_radius = 10
gap_up = 1
xs = gf.cross_section.cross_section(width=waveguide_width, layer=layer)
coupler1 = gf.path.arc(radius=couple_radius, start_angle=-90, angle=90)
coupler1 = c << (coupler1.extrude(xs))
coupler1.dmovey(-couple_radius)
coupler1.dmove(
    [(ring_radius + couple_radius + 0.5 * (ring_width + waveguide_width) + gap_up) * np.cos(135 / 180 * np.pi),
     (ring_radius + couple_radius + 0.5 * (ring_width + waveguide_width) + gap_up) * np.sin(135 / 180 * np.pi)])
coupler2 = gf.path.arc(radius=couple_radius, start_angle=-180, angle=90)
coupler2 = c << (coupler2.extrude(xs))
coupler2.dmovey(-couple_radius)
coupler2.dmove(
    [(ring_radius + couple_radius + 0.5 * (ring_width + waveguide_width) + gap_up) * np.cos(45 / 180 * np.pi),
     (ring_radius + couple_radius + 0.5 * (ring_width + waveguide_width) + gap_up) * np.sin(45 / 180 * np.pi)])
route = gf.routing.route_single(
    c,
    coupler1['o2'],
    coupler2['o1'],
    bend=gf.components.bend_circular,
    radius=-0.005 + 0.5 * abs(coupler1.ports['o2'].dx - coupler2.ports['o1'].dx),
    cross_section=xs,
)
print(route.length)
c.show()

### "print(route.length)" the result shows 10, obviously the path length is much larger than 10, how to get the correct path length?

joamatab, 2024-07-07:
Hi

- first issue seems to be a bug in kfactory, thank your help for reporting this!

- for the second issue, there is a straight_factory that you can pass a cross_section to it, and `bend_factory`. I also added a cross_section option in this PR
https://github.com/gdsfactory/gdsfactory/pull/2956

sebastian-goeldi, 2024-07-07:
https://github.com/gdsfactory/kfactory/commit/d6ef9cb75193f19eba4f3d64542b2de33b883462 should fix the route length. note, this is only the manhattan length not the actual path length

sebastian-goeldi, 2024-07-08:
Never mind about the above comment. It's not actually fixed.

The thing about the optical manhattan rout is, the length is only the length of all straight elements (as of now).

The issue about calculating the lengths of bends is that they will require a calculation. How do you truly calculate a truly useful lengths of bends?

Often you would want something like a path length matched. But to truly do that you would need additional info such as n<sub>eff</sub> etc.

The pure backbone lengths might be useful in some cases, so you can calculate it as follows:

`route.length_straight + route.n_bend90 * route.bend90_radius` (this will only work after an update to the newest kfactory)

Note: This is all in DBU not µm