gdsfactory/gdsfactory #2992: how to merge layers without flatten?

SkyLin1020, 2024-07-19:
I want to merge layers like klayout do as below
[image]
is there any funtion can do this now? if not, can you add this function. sometimes fab do not want flattened gds file but merge is needed.

sebastian-goeldi, 2024-07-19:
`gf.kcl.move(LAYER.X,LAYER.Y)` (the layers must be layerenums or the indexes through `gf.layer(layer, datatype)`

SkyLin1020, 2024-07-21:
Hi @sebastian-goeldi ,

thanks for your reply. I think it does not really solve my problem.
when I merge layers in klayout, it merge all shapes in all cells while cell names and their relationship are kept. with the code you replied, it seems nothing happens.

sebastian-goeldi, 2024-07-21:
Could you specify which option you choose in klayout i the merge? There are three options: flat, cell shapes only, or a semi-hierarchichal.

The example I gave is the second one, i.e. cell shapes only.

SkyLin1020, 2024-07-21:
hi @sebastian-goeldi 
this is the layer->merge window, I choose flat.
[image]
I think it really flat all layers in all cells, the the cells list is not flatten.

sebastian-goeldi, 2024-07-21:
Aha. Yes, that's easy:

`c.flatten(merge=True)`

In the background this first calls `c.flatten()` of the klayout cell and then walks through all layers to merge them.

SkyLin1020, 2024-07-21:
@sebastian-goeldi 
I did this before. but the cell list will be flatten. I do not know why my fab want the cell list, but they want me to keep the cell list in the gds file. is there any method to do this?

sebastian-goeldi, 2024-07-21:
The cell list is flat because you probably do `c.write` or similar. Cells aren't deleted but not saved in case you only have this cell and no reference anymore to the other cells.

You can alleviate this with `gf.kcl.write(...)` or `c.kcl.write` instead of the direct `c.write`

This can come with various other problems but should solve your immediate problem.

sebastian-goeldi, 2024-07-21:
Yes? You can't merge shapes and have the cell refs they are coming from them  be in there as well. That's not how gds (or oasis) works. That's also not how the klayou merge works. If you choose flatten there, it will also remove any cell references to c1/c2 (you can confirm this if you look at the the cell hierarchy in klayout. In klayout unless you rightclick the cell and use `save as...`, you are using the equivalent of `gf.kcl.write`

SkyLin1020, 2024-07-22:
I am misunderstanding. Indeed, klayout will delete the shape in 'c1'/'c2', and generate a new shape in 'c'. but no cell is removed or moved out of 'c'. Wwith flatten, it will remove cell references as you say.  With merge shapes code you give above, no merge shape appear and all cells keep unchanged. The code shows as below

```
class LayerMap0(LayerMap):
    L1: Layer = (37, 1)
    L2: Layer = (37, 2)

LAYER = LayerMap0

for layer in LAYER:
    r = gf.kdb.Region(c.shapes(layer))
    r = r.merge()
    c.clear(layer)
    c.shapes(layer).insert(r)
```

SkyLin1020, 2024-07-22:
@sebastian-goeldi , I think this issue can be closed. I received reply from my fab that they want merged version to do DRC and unmerged/unflatten version to avoid some problems in tape-out. So the request is not necessary.

sebastian-goeldi, 2024-07-22:
Very good. That request from the fabe makes much more sense!

And yes: The code above would be for the second merge behavior of KLayout. It will simple merge if you have for example two overlapping rectangles, you will get one polygon out of it. If a reference overlaps a polygon or two references overlap each other, nothing happens.

closed 2025-02-19 by sebastian-goeldi