gdsfactory/gdsfactory #2997: Unintuitive Boolean Operation Results with Stacked Geometries

Daasein, 2024-07-19:
# Description
When performing a boolean 'not' operation on stacked geometries, the results can be unintuitive. Specifically, when creating three polygons around the origin and then adding two circles for subtraction, the boolean 'not' operation does not produce the expected result.

# Reproduce
```python
import gdsfactory as gf
c1 = gf.Component()
rec1 = c1 << gf.components.rectangle((10, 100), layer=(1, 0))
rec2 = c1 << gf.components.rectangle((100, 10), layer=(1, 0))
points = [
    (0, 0),
    (0.7*10, 0),
    (100,100-0.7*10),
    (100, 100),
    (100-0.7*10, 100),
    (0, 0.7*10),
]
rec3 = c1.add_polygon(points=points, layer=(1, 0))
c1
```
[image]

Add two circles for later subtraction:
```python
c2 = gf.Component()
circle1 = c2 << gf.components.circle(radius=10/3, layer=(2, 0))
circle2 = c2 << gf.components.circle(radius=10/3, layer=(2, 0))
circle2.dmove((100,100))
c1.add_ref(c2)
c1
```
[image]
Apply the boolean 'not' operation:
```python
c3 = gf.Component()
c3 << gf.boolean(c1,c1,layer1=(1,0),layer2=(2,0),operation='not',layer=(1,0))
c3
```
[image]
**Observed Behavior**
* After performing the 'not' operation, the bottom left (origin point) remains covered by layer (1, 0).
* Two polygons are generated as a result, one of them is exactly what we want, and the other one is unwanted.

# Expected Behavior
In Klayout, performing the subtraction (layer 1 - layer 2) results in a single polygon in layer 1, which is more intuitive:
[image]

# Proposed Solution
* Consider applying a boolean 'or' operation to all structures in layers A and B before performing the 'not' operation.
* This would ensure a more intuitive result where no remnants are left in the subtracted region.

joamatab, 2024-07-21:
Sounds good, can you send a Pull Request for this?

Daasein, 2024-07-21:
Hi @joamatab,
I added a pull request [#3007](https://github.com/gdsfactory/gdsfactory/pull/3007), please check it.

joamatab, 2024-07-23:
Thank you Dassein,

fixed by https://github.com/gdsfactory/gdsfactory/pull/3013

closed 2024-07-23 by joamatab