gdsfactory/gdsfactory #3008: gf.functions.get_point_inside is misleading

nikosavola, 2024-07-22:
**Describe the bug**
I used `gf.functions.get_point_inside` thinking it would be something similar to [`representative_point` in Shapely](https://shapely.readthedocs.io/en/stable/reference/shapely.GeometryCollection.html#shapely.GeometryCollection.representative_point). However, it simply returns the first polygon point in a component.

This can cause problems at least in my use case when the polygon start coincides with the ports. Then the point actually lies on the edge between two components. Consequently, we cannot distinguish (in a flattened case) which component, for example, a label at the point belongs to.

**Expected behavior**
`get_point_inside` returns a point inside the component, not possibly on the edge.

**Suggested fix**
Change behaviour to match [`representative_point` in Shapely](https://shapely.readthedocs.io/en/stable/reference/shapely.GeometryCollection.html#shapely.GeometryCollection.representative_point) or return the centroid or similar.

sebastian-goeldi, 2024-07-22:
How would you distinguish components i the flattened case anyways?

nikosavola, 2024-07-23:
> How would you distinguish components in the flattened case anyways?
> 
> For any normal polygon (non-selfintersecting and no 0 area parts) you could also just pick the center of the first edge and move it inside by 1dbu. This would fail in case of having any 0 area triangles inside (triangles as in when decomposing the polygon)

I have separate text labels at the ports which separate component boundaries. Nevertheless I think having something like the Shapely method makes sense in gdsfactory as a helper.

sebastian-goeldi, 2024-07-23:
The problem with the point inside like shapely is just that you are calculating the world to just get one point. Yes you can, but it's a comparatively slow approach (especially for big compinents) as we are getting all shapes.

It's similarly slow, albeit still a factor of 10-100 due to being C++ only, as if you were to use `get_polygon_points`

What you probably want for a fast approach here:

```python
#grab first port and get a point directly behind the port
p = c.ports[0].dcplx_trans * gf.kdb.DPoint(-0.001, 0) #assumes dbu is 0.001
```

This assumes the bbox is not empty, you can check that with `c.bbox().empty()`

nikosavola, 2024-07-23:
> The problem with the point inside like shapely is just that you are calculating the world to just get one point. Yes you can, but it's a comparatively slow approach (especially for big compinents) [...]

Hmm, I guess this approach would work if we can generalize to work move in the direction opposite to the port (or maybe component center?)

sebastian-goeldi, 2024-07-23:
This is already genrralized to give you a point "behind" the port. It actually gives you the um Point one dbu (assuming dbu == 0.001) and then put to the place of the port. The way that transformations work, this is guaranteed to be behind the port. Unless your port is on either a 0 area part  of the polygon or not on any polygon at all.

sebastian-goeldi, 2024-07-23:
This is already genrralized to give you a point "behind" the port. It actually gives you the um Point one dbu (assuming dbu == 0.001) behind and then put to the place of the port. The way that transformations work, this is guaranteed to be behind the port. Unless your port is on either a 0 area part  of the polygon or not on any polygon at all.