gdsfactory/gdsfactory #3021: [Question] Why is LayerMap an enum?

95-martin-orion, 2024-07-23:
**Is your feature request related to a problem? Please describe.**
Prior to v8.0, `LayerMap` was a pydantic `BaseModel` object. This was convenient for me for two key reasons:

1. Typing `MyLayerMap.[tab]` will auto-complete with layer names in most IDEs
2. Multiple `LayerMap`s can be combined with inheritance:

```python
class ChipLayers(LayerMap): {...}
class MarkerLayers(LayerMap): {...}

# inherits all members and allows auto-completion
class AllLayers(ChipLayers, MarkerLayers): {...}
```

Starting in v8.0, `LayerMap` derives from `LayerEnum`. This preserves the auto-completion behavior, but python prohibits inheriting from two enum types with distinct values. All other methods of constructing a combined `LayerMap` lose the auto-completion, aside from writing a script to generate the map and running it whenever changes are made.

**Describe the solution you'd like**
This isn't really a request for a fix so much as a request for clarity: why was this change made? Is there a "standard" way to achieve the combined map described above in v8.0?

sebastian-goeldi, 2024-07-23:
That's because pretty much all klayout internal objects will only accept the layer index. Which is the value of these enums.

Yes, unfortunately you can't merge them.

We are looking into just using the names and then having a function to retrieve the index instead. But it won't be tuples again.

95-martin-orion, 2024-07-24:
Thanks for clarifying! I've been toying around with some workarounds in the meantime:

1. Add all layers to a single `LayerMap` and define methods in that class to return subgroups. This forces all layers to be defined in one place, but it preserves auto-complete and is minimally invasive.
2. Define an abstraction above `LayerMap` that supports auto-complete and combination, and only convert it to `LayerMap` when required by GDSFactory. This is closer to what I was hoping for, but converting back and forth might end up being too invasive to be worth it.