gdsfactory/gdsfactory #3022: Incorrect rectangle size due to `snap_to_grid_2x`

Daasein, 2024-07-23:
**Describe the bug**
When creating rectangle at certain sizes, the true size of rectangle is in correct

**To Reproduce**
If you create a rectangle with a decimal point of .785 in one dimension, the actual size of the rectangle will be 1 nm smaller.
```python
import gdsfactory as gf
c = gf.components.rectangle(size=(3.785,2),layer=(1,0))
c.bbox_np()
# c.bbox_np()=array([[1.000e-03, 0.000e+00],
#                   [3.785e+00, 2.000e+00]])
```
[image]
[image]

**Expected behavior**
Because the decimal precision of the passed size is up to the third place and is not smaller than the size of dbu, the size of the rectangle should be exactly equal to size.

**Reason**
[image]
In `gf.components.compass.py`,  `snap_to_grid2x` is used. When size is passed as `(3.785,2)` , it will be reduced to `(3.784, 2)`.

**Suggested fix**
Use `snap_to_grid` rather than 'snap_to_grid2x', and define the points in `gf.components.compass.py` as:
```python
points = [
        [0,0],
        [dx,0],
        [dx,dy],
        [0,dy],
    ]
```

sebastian-goeldi, 2024-07-24:
This is a snapping problem with ports. If you make the rectangle not 2xdbu sized, your port is offgrid. Since these rectangles have ports, this is not an option.

If you pureluy want a rectangle you can add it as follows:

```python
c.shapes(LAYER.MYLAYER).insert(gf.kdb.DBox(left, bottom, right, top)) # where left, bottom, right, top are coordinates
```

Daasein, 2024-07-24:
Hi, @sebastian-goeldi, I fully understand the snapping problem. I understand the snapping problem. However, don't you think it is too complicated for a user to use the following code just to add a simple rectangle of the correct size?
```python
c.shapes(LAYER.MYLAYER).insert(gf.kdb.DBox(left, bottom, right, top)) # where left, bottom, right, top are coordinates
```

Daasein, 2024-07-24:
Hi @sebastian-goeldi ,

I fully understand the snapping problem. However, don't you think it is too complicated for a user to use the code you mentioned  just to add a simple rectangle of the correct size?

kwonka, 2024-07-24:
Noticed the same issue. It makes sense for centered rectangles to snap to "even" widths and heights if symmetry is the priority, but there is no reason that the same restriction should apply to uncentered rectangles. It's not the biggest deal, but it would be nice if these cases could be handled differently.

sebastian-goeldi, 2024-07-24:
I don't think so, no. To be a fair comparison let's look at the full code:

compass
```python
# define w,h for size
compass = c << gf.compass(size=(3,5), layer=LAYER.WG)
compass.move(x,y)
```
pure klayout:
```python
b = gf.kdb.DBox(3,5)
b.move(x,y)
c.insert(LAYER.WG).insert(b)
```

I would say the difference is negligible, especially considering that you don't want a box with ports (otherwise it's simply impossible to have not 2*dbu multiple sizes).

joamatab, 2024-07-25:
the snap to grid 2x is happening to avoid off-grid ports

closed 2024-10-19 by joamatab