gdsfactory/gdsfactory #3023: Document or revert change in supported arg types

95-martin-orion, 2024-07-23:
**Describe the bug**
As of v8.0, `list` is no longer supported as an argument type for `@gf.cell` functions. This presumably extends to all unhashable types (e.g. `dict`, `np.ndarray`, etc.) and is likely the root cause behind #2861.

**To Reproduce**
```python
@gf.cell
def foo(x: list[int]) -> gf.Component:
    return gf.Component()

k = foo([1,2,3])
# In gdsfactory==7.27.0, this succeeds.
# In gdsfactory==8.0.0, this fails with "TypeError: unhashable type: 'list'
# Also tested at gdsfactory==8.5.4, which behaves like v8.0.0
```

**Expected behavior**
This isn't a bad change, since unhashable (or more specifically _mutable_) types have unsafe cache behavior. The issue is that it's not called out in the [migration guide](https://gdsfactory.github.io/gdsfactory/notebooks/21_migration_guide_7_8.html) and the effects are wide-reaching; anyone who made heavy use of lists, dicts, or other unhashable types in their use of v7.* will require significant effort to convert to hashable arguments, and may not be aware which types are acceptable and which are not.

**Suggested fix**
I can think of a few options, in no particular order:

1. Convert these types in the migration script. This seems incredibly tricky to get right, but done correctly it would completely eliminate this issue.
2. Document the change in the migration guide. This is the easiest option, but may leave some users with a lot of work to do before they can upgrade to v8.*. This might be useful to do in addition to (1), as well.
3. Change `@gf.cell` to support unhashable types again. I can't really recommend this given the aforementioned dangers of caching mutable types, but if there's a good way to avoid those issues this option would require the least work from users.

sebastian-goeldi, 2024-07-23:
This should be fixed in newer kfactory versions (something like 0.18+ if not earlier). 

The way we get around unhashable types is by casting lists to tuples and dicts to frozensets on submission to the cache and back on retrieval.

If this is not the case in any tests, then it should be a bug of kfactory. I thought we have tests, but I couldn't find them. I'll investigate tomorrow.

sebastian-goeldi, 2024-07-25:
I checked. Unfortunately this is not true for lists, that's right. The problem is that we can't do the magic trick of `frozenset` like we could with dicts. I have some ideas how we might be able to solve this, but I need to test first and see whether that works.

joamatab, 2024-11-15:
Hi Orion, we fixed this in Kfactory side

Feel free to reach out if there’s anything else I can assist with.

closed 2024-11-15 by joamatab

95-martin-orion, 2024-11-15:
Thanks! Do you know which gdsfactory version picked up / will pick up the change? I think it's working for me at `v8.17.0` but I haven't checked closely.

sebastian-goeldi, 2024-11-15:
Quite some time ago, so 8.17 should definitely work