gdsfactory/gdsfactory #3036: the temporary oas files in the wrong place

JinWallner, 2024-07-25:
**Describe the bug**
What's the bug?
When use show() for the component created by import_gds, the temporary oas file are generated in the wrong place

**To Reproduce**
What's the code to reproduce the behavior? What commands or code did you write to get the error?
import gdsfactory as gf
import os
import pathlib

gdsname = "./gds/test.gds"
#gdsname = os.path.join(pathlib.Path(__file__).parent.absolute(), "gds", "test.gds")

c = gf.Component("Top")
c << gf.import_gds(gdsname) 
c.show()

**Expected behavior**
What would you like to happen?
If the code run as it s, it will have RuntimeError that unbable to ope the test.oas file for write. The Path for the temporary file is not correct. If uncomment the second line of gdsname definition, the test.oas will be created in the folder where the original gds file was stored instead of the build folder. It's not the desired behavior to add tempory file in PDK folders.

**Suggested fix**
How could we fix the bug?

tweax-sarl, 2024-07-26:
I just stumbled over the same issue. I must admit that I am pretty new to gdsfactory and klayout and I don't understand all the underlying stuff.

In my case I can circumvent the issue by using c.show(use_libraries=False):
```
c = gf.Component("Top")
c << gf.import_gds(gdsname)
c.write_gds('test.gds')   # works as intended
c.show(use_libraries=False)  # fixes the issue for me
```

JinWallner, 2024-07-26:
The use_libraries=False does work. Thank you very much.

sebastian-goeldi, 2024-07-27:
Hmm... Interesting. I am not entire sure why kfactory would not write the libraries to the same location as the file itself (just glanced at the code couldn't figure out what the issue would be), nevertheless there's probably a bug there.

Fix on gdsfactories side should be easy though.

@joamatab if you see this before I make the PR. In `import_gds` call `temp_kcl.delete()` at the end, that should fix it.

joamatab, 2024-07-27:
Hi Jin,

thanks for reporting this!

Sebastian, we can wait for your PR on kfactory side

closed 2024-07-27 by joamatab