gdsfactory/gdsfactory #3040: Flattening a component with a label on the top cell removes the label

nikosavola, 2024-07-26:
**Describe the bug**
Flattening a component with a label on the top cell appears to remove the label.

**To Reproduce**
```python
import gdsfactory as gf

@gf.cell
def flattened_hier_cell_with_top_cell_text() -> gf.Component:

    c = gf.Component()
    _straight1 = gf.components.straight()
    straight1 = c << _straight1

    c.add_label(
        text="Some important text at the port",
        position=straight1.ports["o1"].dcenter,
        layer=straight1.ports["o1"].layer,
    )
    c.show()
    breakpoint()
    c.flatten()
    c.show()

    return c

if __name__ == '__main__':
    c = flattened_hier_cell_with_top_cell_text()
    c.show()
```

**Expected behavior**
The label is included in the final geometry

sebastian-goeldi, 2024-07-26:
This is a bug in kfactory. If `merge=False` is passed it works (because the merging is done with regions and regions swallow labels ;)

joamatab, 2024-10-19:
closing this as it was fixed

closed 2024-10-19 by joamatab