gdsfactory/gdsfactory #3060: [Question] How to place the route between ports on a specific layer?

wagnojunior, 2024-07-30:
Some routing functions, such as `gdsfactory.routing.route_single_electrical`, `gdsfactory.routing.route_quad`, and `gdsfactory.routing.route_bundle`, have a `layer` argument, which makes it very easy to choose on which layer the route is placed.

However, other functions, such as `gdsfactory.routing.route_single`, `gdsfactory.routing.route_single_from_steps`, and `gdsfactory.routing.route_bundle_all_angle`, don't have a `layer` argument. In these cases, how can I place the route on a layer of my choosing?

I suppose that `cross_section` is somehow related to it, but I could not wrap my head around it yet.

joamatab, 2024-08-16:
Hi, you can read more about how to use cross_sections in our tutorial

closed 2024-08-16 by joamatab

wagnojunior, 2024-08-19:
In case anyone else has a similar question, here is how I solved it. PS: I am connection two electrical pads of port `electrical` using 45 degrees wire bend.

1. Create a custom cross section: here, you can control the width and the layer of the cross-section. Once you assign this cross-section to the route function, the width of the wire will be the same as the width of the cross-section, and the layer on which the route is placed is the layer of the cross-section.
```python
cross_section = gf.cross_section.cross_section(
            width=15,
            offset=0,
            layer=(1,0),
            port_types=['electrical', 'electrical']
)
```
2. Create a custom wire corner: if you don't create a custom wire corner, the width of the corner will be different than the width of the wire. In here, the argument `cross_section` should be the cross-section that was created previously.
```python
wire_corner45 = gf.components.wire_corner45(
            width=15,
             cross_section=cross_section
        )
```
3. Route: remember to include the cross-section and the wire corner. The port type of the route should match that of the cross-section.
```python
wire_1_i = gf.routing.route_single(
            component=c,
            port1=port_1,
            port2=port_2,
            bend=wire_corner45,
            taper=None,
            cross_section=cross_section,
            port_type='electrical',
            allow_width_mismatch=True,
        )
```

That is it~!