gdsfactory/gdsfactory #3083: How to remove additional hidden cell named "$$$CONTEXT_INFO$$$"

kazimgorgulu, 2024-08-02:
Hi, `write_gds` function always creates a hidden top cell with the name "$$$CONTEXT_INFO$$$" when I import the gds file with `gdstk`. I tried to remove this by setting the environment variable `write_context_info=False` but it didn't work.

```python
import gdsfactory as gf
gf.CONF.write_context_info=False

gf.clear_cache()
c = gf.Component("test_cell")  
p1 = c.add_polygon([(-1, -6), (6, 8), (7, 17), (9, 5)], layer=(2, 0))
c.write_gds("test.gds")
c2 = gf.import_gds("test.gds")
print(c2.name)

output:
test_cell
```

```python
import gdstk
library = gdstk.read_gds("test.gds")
cells = library.cells
cell_names = [cell.name for cell in cells]
print(cell_names)

output:
['$$$CONTEXT_INFO$$$', 'test_cell']
```

What should I do to not have `$$$CONTEXT_INFO$$$` top cell in the gds file? 
btw I am using gdsfactory v8.

Daasein, 2024-08-02:
I encountered the same issue. I am using the latest version of KLayout, and `$$$CONTEXT_INFO$$$` is not a problem with the latest KLayout. However, when I try to load the .gds file in Beamer or an older version of KLayout (around version 0.25, if I recall correctly), an error occurs. To resolve this, I need to manually "save as" in KLayout and uncheck the "Store PCell and library context information (format specific)" option. This allows Beamer or the older version of KLayout in the cleanroom to read the file without issues.
[image]

Daasein, 2024-08-02:
@kazimgorgulu 
This works for me, please try it:
```python
from kfactory.kcell import save_layout_options
import gdsfactory as gf
c = gf.Component("test_cell")  
p1 = c.add_polygon([(-1, -6), (6, 8), (7, 17), (9, 5)], layer=(2, 0))
options = save_layout_options(write_context_info=False)
c.write_gds(r"dummy.gds", save_options=options)
```

sebastian-goeldi, 2024-08-02:
`c.write_gds(with_metadata=False)` will work as well.

Note, this setting will turn off any port or otherwise used metainfos and library cells (though gdsfactory isn't using these yet). I.e. all ports infromtations or parameters or additional infos will be lost in the written gds/oasis.

joamatab, 2024-08-02:
Thank you for reporting this,

You can do this

```python
import gdsfactory as gf

c = gf.Component("test_cell")
p1 = c.add_polygon([(-1, -6), (6, 8), (7, 17), (9, 5)], layer=(2, 0))
c.write_gds("test.gds", with_metadata=False)
c2 = gf.import_gds("test.gds")
print(c2.name)

```

kazimgorgulu, 2024-08-02:
Thank you all for the replies

closed 2024-08-02 by kazimgorgulu