gdsfactory/gdsfactory #3086: Layer extract() function is slow for multiple layers

octavephotonics, 2024-08-02:
**Describe the bug**
Running the extract() function with multiple layer arguments on a component imported from a gds file takes an unexpectedly long time to execute compared to extracting layers individually. 

**To Reproduce**
Here is a minimal working example to reproduce the issue. The code below imports the attached gds file, extracts Layer 0 and Layer 4 individually, then extracts both layers through a single function call to extract().

```
[demolines.zip](https://github.com/user-attachments/files/16475798/demolines.zip)

import gdsfactory as gf
import time

foo = gf.Component()
chip = gf.read.import_gds('demolines.gds')

tic = time.perf_counter()
foo << chip.extract(layers=(0,0))
toc = time.perf_counter()
print('Layer 0 Time: %f s'%(toc-tic))

tic = time.perf_counter()
foo << chip.extract(layers=(4,0))
toc = time.perf_counter()
print('Layer 4 Time: %f s'%(toc-tic))

tic = time.perf_counter()
foo << chip.extract(layers=((0,0),(4,0)))
toc = time.perf_counter()
print('Layer 0 and 4 Time: %f s'%(toc-tic))
```
Here is typical output from the above code:
```
Layer 0 Time: 0.000266 s
Layer 4 Time: 0.000288 s
Layer 0 and 4 Time: 20.520508 s
```

**Expected behavior**
The run time to import multiple layers should not be significantly longer than the sum of the time it takes to import each individual layer.

joamatab, 2024-08-02:
Hi Octavephotonics,

the code posted has an issue, see corrected version below

```python
import gdsfactory as gf
import time

foo = gf.Component()
chip = gf.read.import_gds('demolines.gds')
chip.show()

tic = time.perf_counter()
c1= chip.extract(layers=[(0,0)])
toc = time.perf_counter()
print('Layer 0 Time: %f s'%(toc-tic))

tic = time.perf_counter()
c2= chip.extract(layers=[(4,0)])
toc = time.perf_counter()
print('Layer 4 Time: %f s'%(toc-tic))

tic = time.perf_counter()
c3= chip.extract(layers=((0,0),(4,0)))
toc = time.perf_counter()
print('Layer 0 and 4 Time: %f s'%(toc-tic))
```

Layer 0 Time: 0.121232 s
Layer 4 Time: 31.529844 s
Layer 0 and 4 Time: 34.504553 s

We can investigate how to make this faster

joamatab, 2024-08-03:
i got to run the code much faster

Layer 0 Time: 0.060577 s
Layer 4 Time: 0.001984 s
Layer 0 and 4 Time: 0.002031 s

This was reported before but i didn't find the issue at that time

I'll release a patch for this to fix it in a new gdsfactory version

@mdecea

octavephotonics, 2024-08-03:
Thanks Joaquin!

closed 2024-08-03 by octavephotonics