gdsfactory/gdsfactory #3088: write_gds() much slower in gdsfactory8 vs gdsfactory7 for polygons generated from points

octavephotonics, 2024-08-03:
**Describe the bug**
Exporting a component to a gds file is >10x slower for some geometries in gdsfactory8 compared to earlier versions. 

**To Reproduce**
The code below generates a polygon comprised of a list of points. In general, the polygon could be some arbitrary shape, but here it is a simple ring. Exporting the component via the export_gds() function is significantly slower (10x or more) in gdsfactory8 compared to earlier versions. Reading the same gds file back in and then copying to a new file by running export_gds() a second time is fast.

```
import gdsfactory as gf
import numpy as np
import time

#Make a polygon (in this example approximating a ring) from a bunch of points
def testRing(total_points, radius, ring_width):
    # Define the angular points for the ring
    theta = np.linspace(0, 2 * np.pi, total_points)
    
    # Inner and outer radius
    inner_radius = radius - ring_width / 2
    outer_radius = radius + ring_width / 2
    
    inner_r = inner_radius * np.ones_like(theta)
    outer_r = outer_radius * np.ones_like(theta)
    
    # Create the points for the inner and outer boundaries of the ring
    inner_points = np.array([inner_r * np.cos(theta), inner_r * np.sin(theta)]).T
    outer_points = np.array([outer_r * np.cos(theta), outer_r * np.sin(theta)]).T
    points = np.concatenate([inner_points, outer_points[::-1], [inner_points[0]]])
    
    # Create the component
    component = gf.Component("test_ring")
    component.add_polygon(points, layer=(1, 0))
    
    return component

# Ring Parameters
radius = 100  # radius of the ring
ring_width = 4  # width of the ring
totalpoints = 100000

# Create the ring
ring = testRing(totalpoints, radius, ring_width)

tic = time.perf_counter()
ring.write_gds('test_ring_from_pts.gds')
toc = time.perf_counter()
print('write_gds() after polygon time: %f' % (toc-tic) )

ring2 = gf.read.import_gds('test_ring_from_pts.gds')
tic = time.perf_counter()
ring2.write_gds('test_ring_from_ptsv2.gds')
toc = time.perf_counter()
print('write_gds() after import time: %f' % (toc-tic) )
```

Example output:
```
write_gds() after polygon time: 37.073882
write_gds() after import time: 0.007212
```

**Expected behavior**
Comparable or better performance for exporting gds files in gdsfactory8 compared to gdsfactory7.

joamatab, 2024-08-03:
Hi Octavephotonics,

I get,

write_gds() after polygon time: 2.191443
write_gds() after import time: 0.006099

sebastian-goeldi, 2024-08-03:
Also, if you leave away the KCell creation ut is likely faster than gdstk. The metadata has some cost to reconstruct ports, settings, info

octavephotonics, 2024-08-03:
Thanks guys. I'll investigate more on Monday. It could be there is some other configuration problem on the test computer since you're unable to reproduce the slow down.

DanHickstein, 2024-08-05:
I can reproduce David's issue. I'm working with a fresh install of Python (3.12.4) and the latest pip version of gdsfactory (8.6.4). 

```
write_gds() after polygon time: 49.307181
write_gds() after import time: 0.011170
```

octavephotonics, 2024-08-06:
We've tried on three different computers (all Windows) with both fresh Anaconda environments and existing environments and see consistent behavior comparing gdsfactory v7 vs v8.

Fresh conda environment with gdsfactory v8.6.7: "pip install gdsfactory"
```
write_gds() after polygon time: 38.595865
write_gds() after import time: 0.007612
```

Fresh conda environment with gdsfactory v7: "pip install gdsfactory==7.27.1"
```
write_gds() after polygon time: 1.581364
write_gds() after import time: 0.008726
```

joamatab, 2024-08-06:
very strange, im getting this times in gdsfactory8

write_gds() after polygon time: 1.790433
write_gds() after import time: 0.005587

can you  append this to the script?

```python
import sys
print(sys.version)
print(sys.executable)
print(gf.config.__version__)
```

octavephotonics, 2024-08-06:
Here is the script output with the version info added:
```
3.11.7 | packaged by Anaconda, Inc. | (main, Dec 15 2023, 18:05:47) [MSC v.1916 64 bit (AMD64)]
C:\Users\DavidCarlson\anaconda3\python.exe
8.6.7
write_gds() after polygon time: 39.908851
write_gds() after import time: 0.006780
```

DanHickstein, 2024-08-06:
I changed a few variables at once, but here is Python 3.10 on a Mac. It runs fast (and has some additional warnings):

```
3.10.9 (main, Mar  1 2023, 12:20:14) [Clang 14.0.6 ]
/Users/danielhickstein/anaconda3/bin/python
8.6.7
write_gds() after polygon time: 0.812974
Warning: Record length larger than 0x8000 encountered: interpreting as unsigned (position=124, record number=9, cell=test_ring)
Warning: Record length larger than 0x8000 encountered: interpreting as unsigned (position=54068, record number=14, cell=test_ring)
Warning: Record length larger than 0x8000 encountered: interpreting as unsigned (position=102996, record number=19, cell=test_ring)
Warning: Record length larger than 0x8000 encountered: interpreting as unsigned (position=145940, record number=24, cell=test_ring)
Warning: Record length larger than 0x8000 encountered: interpreting as unsigned (position=181228, record number=29, cell=test_ring)
Warning: Record length larger than 0x8000 encountered: interpreting as unsigned (position=273932, record number=39, cell=test_ring)
<36 similar warnings deleted...>
Warning: Record length larger than 0x8000 encountered: interpreting as unsigned (position=1038036, record number=119, cell=test_ring)
write_gds() after import time: 0.004504
```

joamatab, 2024-08-06:
im also seeing those warnings on linux

im not sure why it's so slow on windows

octavephotonics, 2024-08-07:
FWIW I tried saving to OASIS as a possible workaround but it's still just as slow. 

Is the conclusion that this is a problem with Klayout's Windows implementation?

GCShao, 2024-09-11:
I encountered a similar scenario!!!! I couldn't believe it's getting worser!

sebastian-goeldi, 2024-09-12:
The reason this is slow on first write and fast on read/write is because in the first case, the polygon needs to be split into gds compliant polygons (8192 points maximum when cheating and using an unsigned integer, 4096 with standard signed integer). If you were to do

```python
ring2 = gf.read.import_gds('test_ring_from_pts.gds')
r = gf.kdb.Region(ring2.shapes(gf.kcl.layer(1,0)))
r.merge()
ring2.shapes(gf.kcl.layer(1,0)).clear()
ring2.shapes(gf.kcl.layer(1,0)).insert(r)
tic = time.perf_counter()
ring2.write_gds('test_ring_from_ptsv2.gds')
toc = time.perf_counter()
print('write_gds() after import time: %f' % (toc-tic) )
```

You would get the same time again.

P.S. @GCShao if you write issues like this, it will make people not want to respond or help you with issues. After all this is an open source project that we enhance and improve in our free time. We try to make it as fast and efficient as possible, but sometimes we have other more important stuff to do. And sometimes there are reasons why stuff doesn't stay the same. We moved to klayout as a backend because we thought it's more powerful and better maintained than gdstk. So, yes some stuff is currently not in ideal shape, but a lot of stuff is now better than it has ever been before.

DanHickstein, 2024-09-12:
> After all this is an open source project that we enhance and improve in our free time.

Huge thanks to the gdsfactory team for all of the great work that they do! We're happy to help find a few bugs that crop up along the way. It's the cost of progress!

GCShao, 2024-09-12:
> The reason this is slow on first write and fast on read/write is because in the first case, the polygon needs to be split into gds compliant polygons (8192 points maximum when cheating and using an unsigned integer, 4096 with standard signed integer). If you were to do
> 
> ```python
> ring2 = gf.read.import_gds('test_ring_from_pts.gds')
> r = gf.kdb.Region(ring2.shapes(gf.kcl.layer(1,0)))
> r.merge()
> ring2.shapes(gf.kcl.layer(1,0)).clear()
> ring2.shapes(gf.kcl.layer(1,0)).insert(r)
> tic = time.perf_counter()
> ring2.write_gds('test_ring_from_ptsv2.gds')
> toc = time.perf_counter()
> print('write_gds() after import time: %f' % (toc-tic) )
> ```
> 
> You would get the same time again.
> 
> P.S. @GCShao if you write issues/comments like this, it will make people not want to respond or help you with issues. After all this is an open source project that we enhance and improve in our free time. We try to make it as fast and efficient as possible, but sometimes we have other more important stuff to do. And sometimes there are reasons why stuff doesn't stay the same is because the software evolves. We moved to klayout as a backend because we thought it's more powerful and better maintained than gdstk. So, yes some stuff is currently not in ideal shape, but a lot of stuff is now better than it has ever been before.

Apologies for my words. I am not a gdsfactory expert and might get too emotional when shifting my project to the latest version. I have already sticked my project with the stable environment and it works perfectly. I have to admit this package is the best layout package I have ever used. I will also improve myself and try to make my comments more meaningful and professional!

sebastian-goeldi, 2024-09-12:
No hard feelings, I get it, tapeouts can be super stressful. We are trying our best to make it as stable as possible.

If you aren't a power user (or rather anyone participating in the tapeout), I can highly recommend that you use `pip freeze > requirements.txt`. This will make sure every package is installed exactly with the version when you ran this command.

Also, if you need timely help, I can recommend first opening an issue / discussion on github and then write in the gitter/matrix channel. I (and probably most devs) see that faster than issues on here, which can take a few days until we get to it. On the channel you have more people that read it than purely going through issues.

Anyways, the issue in the example of this is an issue in KLayout's write in the python module. As soon as that one gets fixed, the issue will get fixed in gdsfactorty.

As a workaround you can use the workaround mentioned in the KLayout issue, but it would require to falling back to klayout/kfactory syntax:

```python
# workaround for
# component.add_polygon(points, layer=(1, 0))
dpolygon = gf.kdb.DPolygon([gf.kdb.DPolygon(x, y) for x, y in outer_points])
dpolygon.insert_hole([gf.kdb.DPoint(x, y) for x, y in inner_points])
split_polygons = dpolygon.break_(4000, 0) # splits the polygon into chunks which have a maximum of 4000 points each)
for poly in split_polygons:
    c.add_polygon(poly, layer=(1, 0))
```

This is about as fast on windows as on linux (even with only 2 cores)

sebastian-goeldi, 2024-09-27:
This will be fixed as soon as gdsfactory releases a version with kfactory 0.20.6

sebastian-goeldi, 2024-09-29:
Afaik, this should be fixed with 8.8.9. Can't confirm though, as I will only be back home in Switzerland in a week. If someone else could confirm or deny, then we can close the issue or find out why it didn't work yet.

closed 2024-11-15 by joamatab