gdsfactory/gdsfactory #3102: Bring Back get_bundle_from_steps to route_bundle_from_steps

aerogeorge, 2024-08-07:
Is there a way to do a route bundle from steps in the new v8? I have looked through and there is only a single route version for the steps.

joamatab, 2024-08-09:
not supported yet, PRs are always welcome :)

aerogeorge, 2024-08-09:
hehe... I wish I was code gEnIuS enough to PR

joamatab, 2024-08-09:
we all start somewhere. Keep at it, and you'll get there! 😄

sebastian-goeldi, 2024-08-09:
This is on my plan for kfactory. And as gdsfactory uses kfactory's routing for these cases, at the very least the bundle one, the PR will be there. But I currently have some higher priority items from my employer which I need to finish first, so it will be a while.

The single case should be relatively easy, there is already a ManhattanRouter objecr in the back which essentially allows most of the functionality, so it will just need a wrapper class to allow the functionality to use in the optical case.

For bundle routing it's a bit harder. The steps do not necessarily make sense as they did in 7, especially if you have more than one start direction in your bundle. It would be really helpful if you (or anybody else) could outline what you expect from such a feature.

aerogeorge, 2024-08-11:
Thanks Sebastian.. makes sense and looking forward to the functionality.  I used 7's get_bundle_from_steps feature to fanout rather complicated routes from 100's of regularly spaced diodes to optical and electrical connections. Since there are so many connections, the "steps" feature made it nice to play with the routing and figure out the best, most space efficient routing path. Without this functionality, the alternative way to do this is place anchors along the way at set dx and dy and then route bundle between all of the anchors. I suppose that this would be one of the ways to code in the functionality for a hypothetical "route_bundle_from_steps".

joamatab, 2024-10-15:
closing this as we added this in 8.11.0

https://gdsfactory.github.io/gdsfactory/changelog.html

closed 2024-10-15 by joamatab

sebastian-goeldi, 2024-10-15:
Steps are still coming (though it might be slightly different).