gdsfactory/gdsfactory #3106: @gf.cell lose the

rgiroud, 2024-08-08:
Hi, 

I've come across something when trying to update from gdsfactory 7 -> 8, which I don't remember happening in 7.*

The decorator @gf.cell doesn't seem to provide the docstring of a Component generating function anymore. The Pylance rendering in VSCode is also impacted.

Picture without the decorator: 
[image]

Picture with the decorator: 
[image]

Is this something fixable? 
I tried a few things (without touching gdsfactory code) like implementing my own decorator using functools.wraps which would apply @gf.cell to the decorated function. This solved the rendering and docstring problem, but rose a cell naming problem.

Thanks

sebastian-goeldi, 2024-08-08:
I think the problem is this:
https://github.com/gdsfactory/gdsfactory/blob/main/gdsfactory/cell.py#L69

This will not create a valid overload signature, therefore the `@overload` will not be able to be resolved by pylance (and probably pretty much any other tool).

sebastian-goeldi, 2024-08-08:
I think this might require a small fix on kfactory's side

sebastian-goeldi, 2024-08-20:
The problem with your approach is that it will disallow any settings on the decorator directly.

Wrt layout_cache: This one will first look into the layout object whether a cell with the targeted name is present and return that instead of executing the function. This is likely because you mix using string with crossections or pass multiple partials etc.

joamatab, 2024-11-15:
Hi Robin,

Is this issue still unresolved? If so, I’ll go ahead and close it. Feel free to reach out if there’s anything else we can assist with.

Best regards,

closed 2024-11-15 by joamatab

rgiroud, 2024-11-26:
Hi,

I just updated to gdsfactory to 7.20.0. From what I could observe the @gf.cell decorator still has the same behaviour than described above.

That's not much of an issue tho, I've been doing with my temporary fix in the meantime and if I really needed specific settings from the original decorator, I simply used it instead.