gdsfactory/gdsfactory #3109: Is klayout_pyxs still supported?

m-mcgivney, 2024-08-09:
**Describe the bug**
Is klayout_pyxs still supported?  I am trying to use klayout_pyxs (for the first time) and I have Klayout v0.29.4.

I installed klayout_pyxs using "pip install klayout_pyxs" and installed the package in KLayout using the Salt Package Manager, but no "pyxs" option has appeared in the KLayout -> Tools dropdown menu as I was led to expect based on this page:  https://gdsfactory.github.io/klayout_pyxs/

Can someone please confirm if klayout_pyxs is still supported, and whether I missed a step in the installation process?

joamatab, 2024-08-09:
Hi, 

i think some of the latest klayout updates broke the package

[image]

Feel free to send a PR if you find the fix
https://github.com/gdsfactory/klayout_pyxs/

@dimapu
@klayoutmatthias
@sebastian-goeldi

m-mcgivney, 2024-08-09:
Got it, thanks!

sebastian-goeldi, 2024-08-09:
I think this might be related to the implicit conversions of Polygon, e.g. if you pass it the constructor to a Box or list of points etc.

klayoutmatthias, 2024-08-10:
How can I reproduce the problem?

joamatab, 2024-08-11:
- you can follow the intructions [here](https://gdsfactory.github.io/gdsfactory/notebooks/03_layer_stack.html#klayout-cross-section)

joamatab, 2024-09-11:
I removed the klayout_pyxs from the recommended plugins until we fix it

joamatab, 2024-10-19:
i moved the issue [here](https://github.com/gdsfactory/klayout_pyxs/issues/11) as this is more of a klayout issue

closed 2024-10-19 by joamatab