gdsfactory/gdsfactory #3122: Error in bbox calculation

95-martin-orion, 2024-08-12:
**Describe the bug**
Non-manhattan rotations in subcomponents lead to incorrect values in `bbox_np`. It appears that the subcomponent's rotated bounding box contributes to the total instead of the subcomponent's geometry.

**To Reproduce**

```python
import gdsfactory as gf
import numpy as np

@gf.cell
def pyramid():
    comp = gf.Component()
    comp.add_polygon(((0, 0), (1, 1), (2, 0)), layer=(1, 0))
    return comp

def test_off_axis_bbox():
    comp = gf.Component()
    subcomp = comp.add_ref(pyramid())
    subcomp.drotate(45, center=(0, 0))

    rt2 = np.sqrt(2)
    # The triangle should form the upper-left corner of a rt2-width square.
    assert np.allclose(comp.bbox_np(), ((0, 0), (rt2, rt2)))
    # Error: ((-rt2 / 2, 0), (rt2,  rt2 * 1.5)) != ((0, 0), (rt2, rt2))
```

**Expected behavior**
The bounding box should always be calculated directly from geometry, not from the bounding boxes of child components.

95-martin-orion, 2024-08-12:
The same appears to be true for the `gf.kdb.DBox` produced by `comp.dbbox()`.

sebastian-goeldi, 2024-08-13:
I would argue this is not a bug. It's purely how you define a bbox.

Also, to my knowledge, no fab I know of really allows you to have instances like this (or off-grid instances in general)

If you want the true bbox, flatten the ref in, or use the various tools like recursive shape iterators. Note though, that will be extremely expensive compared to the standard bbox calculation.
 If you still think this is absolutely a bug, I would suggest to open an issue on klayout as the behavior is fully inherited from klayout.