gdsfactory/gdsfactory #3127: gf.component.array for any cell

oz99, 2024-08-14:
I'm attempting to create an array of a custom component that I made which is defined as a cell. 

I'm unsure how to go about making an array efficiently. 

I know I could create a loop to create the desired array but I thought initially that this would already be included into the gf.component.array() function. While this is a component command, I thought it would have the same versatility as the gf.boolean where we can conduct "xor", "and" and "or" functions to any cells in the layout. 

I would be able to help with the base code but would need a bit of guidance.

sebastian-goeldi, 2024-08-14:
Have a look at `Component.add_array` ;)

Same as with the boolean, the function you mention is the legacy support to keep the API from v7

joamatab, 2024-10-19:
closing this for now,

you can also use `Component.ref(component_to_reference, columns=1, rows=3)`

closed 2024-10-19 by joamatab