gdsfactory/gdsfactory #3141: Runtime Error when same routing cell are used in multiple subcells

JinWallner, 2024-08-19:
**Describe the bug**
I had RuntimeError: The following cell name(s) are used for more than one cell - can't write this layout: with the following code. Somehow the routing cells can't be used in more than one cells. If I remove the route part, the code will run. And the same MMI will show up in both sub cell c1 and c2.

**To Reproduce**
def sub_c(
        label='default', 
        )->Component:

    c = gf.Component(label)
    mmi1 = c << gf.components.mmi1x2()
    mmi2 = c << gf.components.mmi1x2()
    mmi2.dmove((100, 50))
    xs= partial(gf.cross_section.cross_section, layer=(2,0), width=0.6)
    route = gf.routing.route_single(
        c,
        port1=mmi1.ports["o2"],
        port2=mmi2.ports["o1"],
        cross_section=xs,
        route_width=0.6,
    )
    c.add_label(text=f"{label}", position=(0,0), layer=(1,0))
    return c

c = gf.Component("Top")
sub1 = c << sub_c("c1")
sub2 = c << sub_c("c2")
sub2.dmove((0, 100))
c.show()

**Expected behavior**
What would you like to happen?

**Suggested fix**
How could we fix the bug? As an open-source project, we welcome your suggestions and PRs.

**Environment (please complete the following information):**

- [x ] I have checked the documentation and found no answer
- [ x] I have checked the issues and found no answer
- [ x] Im using the latest GDSFactory version and python 3.10, 3.11 or 3.12. Please copy below the output for this code:

```python
import sys
print(sys.version)
print(sys.executable)

import gdsfactory as gf
gf.config.print_version_plugins()
```

sebastian-goeldi, 2024-08-19:
Hi Jin,

The problem is the partial of the crosssection in your code. The partial is hashable, but the two calls will result in a different hash and therefore the second call will be a cache miss for any straights/bends created in the routing.

It is sufficient to take the cross_section definition outside as I see no problem doing so in your example. Should it be mroe dynamic, you need to make sure the same object is created for the same args, e.g. through `get_cross_section` (I thin that's a thing if I followed Joaquin's code enough).

Working example:

```python
import gdsfactory as gf
from functools import partial

xs = partial(gf.cross_section.cross_section, layer=(2, 0), width=0.6)

def sub_c(
    label: str = "default",
) -> gf.Component:
    c = gf.Component(label)
    mmi1 = c << gf.components.mmi1x2()
    mmi2 = c << gf.components.mmi1x2()
    mmi2.dmove((100, 50))
    route = gf.routing.route_single(
        c,
        port1=mmi1.ports["o2"],
        port2=mmi2.ports["o1"],
        cross_section=xs,
        route_width=0.6,
    )
    c.add_label(text=f"{label}", position=(0, 0), layer=(1, 0))
    return c

c = gf.Component("Top")
sub1 = c << sub_c("c1")
sub2 = c << sub_c("c2")
sub2.dmove((0, 100))
c.show()
```

sebastian-goeldi, 2024-08-20:
Hmmm, I suspect there is some weird interaction with rounding going on. I think by default gdsfactory does some rounding on some values in cross sections. @joamatab  is more familiar with it though.

sebastian-goeldi, 2024-08-20:
@joamatab there is an error with `route_single` and the cross_sections. For some reason kwargs are sent with a fully assembled cross section

```pycon
  /home/sgoeldi/miniforge3/envs/gf/lib/python3.12/bdb.py(600)run()
-> exec(cmd, globals, locals)
  <string>(1)<module>()
  /home/sgoeldi/repos/gdsfactory/gdsfactory/test.py(28)<module>()
-> sub1 = c << sub_c("c1")
  /home/sgoeldi/repos/gdsfactory/gdsfactory/test.py(16)sub_c()
-> route = gf.routing.route_single(
  /home/sgoeldi/repos/gdsfactory/gdsfactory/routing/route_single.py(153)route_single()
-> return route(
  /home/sgoeldi/miniforge3/envs/gf/lib/python3.12/site-packages/loguru/_logger.py(1277)catch_wrapper()
-> return function(*args, **kwargs)
  /home/sgoeldi/miniforge3/envs/gf/lib/python3.12/site-packages/kfactory/routing/optical.py(472)route()
-> route = place90(
  /home/sgoeldi/miniforge3/envs/gf/lib/python3.12/site-packages/kfactory/routing/optical.py(1007)place90()
-> wg = c << straight_factory(width=w, length=length)
  /home/sgoeldi/repos/gdsfactory/gdsfactory/routing/route_single.py(120)straight_dbu()
-> return gf.get_component(
  /home/sgoeldi/repos/gdsfactory/gdsfactory/pdk.py(601)get_component()
-> return get_active_pdk().get_component(component, settings=settings, **kwargs)
  /home/sgoeldi/repos/gdsfactory/gdsfactory/pdk.py(331)get_component()
-> return self._get_component(
  /home/sgoeldi/repos/gdsfactory/gdsfactory/pdk.py(371)_get_component()
-> return component(**kwargs)
  /home/sgoeldi/miniforge3/envs/gf/lib/python3.12/site-packages/kfactory/kcell.py(3400)wrapper_autocell()
-> _cell = wrapped_cell(**params)
  /home/sgoeldi/miniforge3/envs/gf/lib/python3.12/site-packages/loguru/_logger.py(1277)catch_wrapper()
-> return function(*args, **kwargs)
  /home/sgoeldi/miniforge3/envs/gf/lib/python3.12/site-packages/cachetools/__init__.py(741)wrapper()
-> v = func(*args, **kwargs)
  /home/sgoeldi/miniforge3/envs/gf/lib/python3.12/site-packages/kfactory/kcell.py(3267)wrapped_cell()
-> cell = f(**params)
  /home/sgoeldi/repos/gdsfactory/gdsfactory/components/straight.py(30)straight()
-> x = gf.get_cross_section(cross_section, **kwargs) # <-----------------------------  this here is the problem
  /home/sgoeldi/repos/gdsfactory/gdsfactory/pdk.py(611)get_cross_section()
-> return get_active_pdk().get_cross_section(cross_section, **kwargs)
> /home/sgoeldi/repos/gdsfactory/gdsfactory/pdk.py(425)get_cross_section()
-> warnings.warn(
```

closed 2024-08-26 by joamatab