gdsfactory/gdsfactory #3142: Port.layer returns an integer

nikosavola, 2024-08-21:
**Describe the bug**
Port.layer returns an int to the KLayout layer instead of `gf.typings.Layer`. This seems a bit funny.

**To Reproduce**
For example,
```python
c = gf.c.straight()
assert c.ports['o1'].layer in c.layers  # Returns false as c.layers returns a list of tuples
```

**Expected behavior**
Port.layer returns same kind of object as other layers

nikosavola, 2024-08-21:
> It's not an integer, it's a LayerEnum. As long as you use the built-in layermap, you can use `tuple(port.layer)` and it should return that. In the newer kfactory I would suggest to use `port.layer_info` instead which return a `gf.kdb.LayerInfo`

In this example `tuple(port.layer)` doesn't appear to work. It has been downcasted to an actual `int` at some point.

sebastian-goeldi, 2024-08-21:
Then you can always go the route of `port.kcl.get_info(port.layer)` that should return you what `port.layer_info` is doing in the new kfactory

sebastian-goeldi, 2024-08-21:
But `port.layer` kind of needs to be an integer as otherwise all the internal stuff like `c.shapes(port.layer)` etc will not work. Theoretically this could be changed on the kfactory side. But then again, a) my company has been using kfactory actively for the last 1+ year, b) tuples are a really bad representation of a layer imo. A model or class makes way more sense imo. Or even just a string.

closed 2025-02-19 by sebastian-goeldi