gdsfactory/gdsfactory #3146: Question: Roughness in euler_bend_all_angle

Nick-Lewis6, 2024-08-22:
Hello,

I'm using the euler_bend_all_angle component to make pulley coupled rings and I noticed that the bends have a roughness on the order of 1nm. Is there any way to fix this? I've attached an image.

Thanks,
Nick

[image]

sebastian-goeldi, 2024-08-22:
Without any further information it's hard to give proper advice. Could you give the settings you are passing to the function?

This definitely doesn't happen on default settings. I guess you are using a very high point density or similar. And yes, there are ways to change this, but without even knowing what your expectations as to how it look like, it's nearly impossible to advise properly.

Also, it's worth pointing out that in pretty much any process this will not have any impact on the final mask.

Butt  the fact that you have so many points in the short section you are showing, I am betting on too high point density.

joamatab, 2024-08-22:
Hi,

Fabrication tolerances at foundries typically range from 1 to 20 nm. These variations arise from the fabrication process itself, rather than the mask resolution.

Nick-Lewis6, 2024-08-23:
Hello,

Thank you for the responses. I understand that 1 nm isn't within fabrication tolerances, but my concern is that this could contribute to sidewall roughness.

For these euler bends I am imputing the bend radius, angle, and width without specifying the number of points. Also, I'm sweeping these three parameters and I've noticed that not all my bends show this roughness. 

Here is how I define my euler bend:
**bend_euler_all_angle(radius = bus_radius, angle = pulley_angle/8, width = wg_w )**

Here's an example of a set of parameters that produce the roughness:
wg_w = **450 nm**
bus_radius = **21.325 um**
pulley_angle/8 = **1 degree** 

Here's an example of a set of parameters that don't produce the roughness:
wg_w = **530 nm**
bus_radius = **24.465 um**
pulley_angle/8 = **11.5 degrees**

joamatab, 2024-08-23:
From the picture you showed your bend seems over-sampled, so i would play with reducing the number of points

sebastian-goeldi, 2024-08-23:
This must be a problem in the `euler` function. If you get a euler bend with 1° angle,path extrude tries to extrude a path with 719 points!

```python
import gdsfactory as gf

gf.components.bend_euler_all_angle(radius=21.325, angle=1, width=0.530).show()
```

```pycon
gdsfactory/gdsfactory/path.py(805)extrude()
-> from gdsfactory.pdk import (
(Pdb) p p
Path(start_angle=0, end_angle=1, points=[[0.00000000e+00 0.00000000e+00]
 [5.16935977e-04 1.30530018e-11]
 [1.03387195e-03 1.04424014e-10]
 ...
 [3.71138853e-01 3.22985734e-03]
 [3.71655710e-01 3.23887903e-03]
 [3.72172567e-01 3.24790079e-03]])
(Pdb) p len(p)
719
(Pdb) 
```

Nick-Lewis6, 2024-08-23:
I understand. I'll try reducing the number of points. Thank you for checking that!