gdsfactory/gdsfactory #3157: RuntimeError: Cannot write layer numbers larger than 65535 to GDS2 streams

dboriska, 2024-08-28:
**Describe the bug**

The write_gds function doesn't work.
Simple  code 
`import gdsfactory as gf`
`c = gf.components.grating_coupler_loss_fiber_array4(pitch=127)`
`c.show()`
gives the error
**RuntimeError: Cannot write layer numbers larger than 65535 to GDS2 streams,**

**Environment**
Fresh installation of Anaconda on Win10
>>> print(sys.version)
**3.12.3** | packaged by conda-forge | (main, Apr 15 2024, 18:20:11) [MSC v.1938 64 bit (AMD64)]

[image]

joamatab, 2024-08-28:
Hi,

can you run `pip install gdsfactory --upgrade` and try again?

dboriska, 2024-08-28:
problem solved after the upgrade

closed 2024-08-28 by dboriska