gdsfactory/gdsfactory #3160: remap_layers does not remap layers

mattlday, 2024-08-29:
**Describe the bug**
After upgrading from GDSfactory 6 to 8, remap_layers no longer works as expected

**To Reproduce**
```
import os
import numpy as np
import gdsfactory as gf
from tkinter import filedialog

file_select=filedialog.askopenfilename()

layout=gf.import_gds(file_select,cellname='TOP')

layers_to_keep = {(304, 100)}  

# Get all layers in the layout
all_layers = set(layout.layers)

# Layers to remove are all layers except those we want to keep
layers_to_remove = all_layers - layers_to_keep

# Remove unwanted layers
layout = layout.remove_layers(layers_to_remove)

# Create rename map
rename_map = {layer: (int(str(layer[0])[0]), 0) for layer in layers_to_keep}

# Remap layers
layout.remap_layers(rename_map)

print("Layers after remapping:", layout.layers)
```

**Expected behavior**
The layer number and datatype should change.

**Suggested fix**

**Environment (please complete the following information):**

- [ Y] I have checked the documentation and found no answer
- [ Y] I have checked the issues and found no answer
- [Y ] Im using the latest GDSFactory version and python 3.10, 3.11 or 3.12. Please copy below the output for this code:

  python │ 3.10.14 
 gdsfactory │ 8.8.1

joamatab, 2024-08-30:
Hi, this is the correct script

```python
import gdsfactory as gf
from tkinter import filedialog

c = gf.components.straight(cross_section='rib_bbox').dup()
c.name = 'TOP'
c.write_gds('straight.gds')
file_select = filedialog.askopenfilename()

layout = gf.import_gds(file_select, cellname="TOP")

layers_to_keep = {(1, 0)}

# Get all layers in the layout
all_layers = set(layout.layers)

# Layers to remove are all layers except those we want to keep
layers_to_remove = all_layers - layers_to_keep

# Remove unwanted layers
layout = layout.remove_layers(layers_to_remove)

# Create rename map
rename_map = {layer: (int(str(layer[0])[0]), 0) for layer in layers_to_keep}

# Remap layers
layout = layout.remap_layers(rename_map) # Remap layers returns a new layout with the layers remapped

print("Layers after remapping:", layout.layers)
```

mattlday, 2024-08-30:
Hey, I tried that but also doesn't work - have you confirmed your script works?

joamatab, 2024-08-30:
it worked for me

i think you need to make sure layers are a list of tuples

[(1,0), (2,0) ]

a layer in GDS has two numbers

closed 2024-10-19 by joamatab