gdsfactory/gdsfactory #3179: get_cells looks at cell decorated functions instead of type annotations

joamatab, 2024-09-03:


MatthewMckee4, 2024-12-28:
This looks at the type annotation of the function
```python
        if not ignore_non_decorated:
            r = signature(func).return_annotation
            return r == Component or (isinstance(r, str) and r.endswith("Component"))
```

joamatab, 2024-12-28:
Sorry, the issue wasn’t entirely clear.

I was considering that, in many cases, we have containers or auxiliary functions that return a component but are not necessarily PCells. Perhaps we should rely on a decorator to explicitly register only the cells we want to expose.

@tvt173 
@flaport 
@sebastian-goeldi

sebastian-goeldi, 2024-12-28:
You can already do that. All `@cell` will be in `kcl.factories`

joamatab, 2024-12-29:
is this a global register?

How do we separate the cells for a particular pdk?

sebastian-goeldi, 2024-12-29:
since it's kcl, not it's not global. As gdsfactory only uses one kcl, well, you don't separate

joamatab, 2024-12-29:
i see, maybe we should add a kcl per PDK for this but that would require the cell decorator to be inside the PDK class

sebastian-goeldi, 2024-12-29:
kfactory already has that, the cell decorator is per kcl

flaport, 2024-12-29:
> Sorry, the issue wasn’t entirely clear.
>
> I was considering that, in many cases, we have containers or auxiliary functions that return a component but are not necessarily PCells. Perhaps we should [...]

That's exactly what get_cells does. With the default arguments it only gets the cells that are decorated with @cell.

flaport, 2024-12-29:
Containers should not be decorated with cell and should hence not be registered as a cell in the PDKs. So it's good that (with default arguments) those won't be retrieved with get_cells

joamatab, 2024-12-29:
sounds good to me, shall i close this now then?

Also, one more question,
What would be the way to access containers in the YAML flow?