gdsfactory/gdsfactory #3180: import gds error: borrowing component from another PDK

Yellowshiba, 2024-09-04:
I need to use some passive components from the ubcpdk (because AppliedNanotool doesn't have a gdsfactory pdk yet) but when I use c.show() to visualize the item in Klayout. The following  error pop up:

[image]

I thought it is because the gds has some specific layers that has large number, so I downloaded the gds of the component and
manually deleted all the layers except the 1/0. And tried to use gdsfactory.read.import_gds to import locally, but I got the same error

The following code reproduce the error, I guess this is not a bug but just I didn't do it correctly. Since I didn't find a good example, I am posting this issue here looking for help.
```python

import gdsfactory as gf
import numpy as np
import matplotlib.pyplot as plt
import scipy.integrate as integrate
from gdsfactory.read import import_gds
import ubcpdk
import ubcpdk.components as uc
from ubcpdk.tech import LAYER
#%%
def DC_geo(L = 30, g = 0.07, R = 5, wg = 3, L_taper = 20):
    c = gf.Component()  # Create a blank component (essentially an empty GDS cell with some special features)
    lda0 = 1.55
    '''
    inf = L/2
    vertices_top = [
        (-inf, g/2 + wg),
        (inf, g/2 + wg),
        (inf, g/2),
        (-inf, g/2),
        
    ]
    ring_wg = c.add_polygon(vertices_top, layer=(1, 0))
    '''
    vertices_bot = [
        (-L/2- L_taper - lda0, -g/2 - wg/2 + 0.25),
        (-L/2 - L_taper, -wg/2 - g/2 + 0.25),
        (-L/2, -g/2),
        (L/2, -g/2),
        (L/2 + L_taper, -g/2 - wg/2 + 0.25),
        (L/2 + L_taper + lda0, -wg/2 - g/2 + 0.25),
        (L/2 + L_taper + lda0, -wg/2 - g/2 - 0.25),
        (L/2 + L_taper, -g/2 - wg/2 - 0.25),
        (L/2, - g/2 - wg),
        (-L/2, -g/2 - wg),
        (-L/2 - L_taper, -g/2 - wg/2 - 0.25),
        (-L/2 - L_taper - lda0, -g/2 - wg/2 -0.25),
    ]
    c.add_polygon(vertices_bot, layer=(1, 0))
    c.add_port(
    name="o1", center=[-L/2- L_taper - lda0, -g/2 - wg/2], width=0.5, orientation=180, layer=(1, 0)
    )
    c.add_port(
    name="o2", center=[L/2 + L_taper + lda0, -g/2 - wg/2], width=0.5, orientation=0, layer=(1, 0)
    )
    return c
#%%
def coupler(L = 30, g = 0.07, R = 5, wg = 3, L_taper = 20):
    c = gf.Component()
    lda0 = 1.55
    dc = c << DC_geo(L = L, g = g, R = R, wg = wg, L_taper = L_taper)
    bl = c <<  gf.components.bend_circular(angle=-90, cross_section='strip', allow_min_radius_violation=False)
    br = c <<  gf.components.bend_circular(angle=-90, cross_section='strip', allow_min_radius_violation=False)
    #gc_in = c << ubcpdk.components.gc_te1550()
    #gc_out = c << ubcpdk.components.gc_te1550()

    bl.connect("o2", dc.ports["o1"], allow_layer_mismatch=True)
    br.connect("o1", dc.ports["o2"], allow_layer_mismatch=True)
    #gc_in.connect("o1", bl.ports["o1"], allow_layer_mismatch=True)
    #gc_out.connect("o1", br.ports["o2"], allow_layer_mismatch=True)
    c.add_port(
    name="o1", center=[-L/2- L_taper - lda0 - R, -g/2 - wg/2 - R], width=0.5, orientation=180, layer=(1, 0)
    )
    c.add_port(
    name="o2", center=[L/2 + L_taper + lda0 + R, -g/2 - wg/2 - R], width=0.5, orientation=0, layer=(1, 0)
    )
    c.plot()
    return c

#%%
L = 260 #length pf the coupling region
g = 0.24
R = 5
wg_w = 3
L_taper = 40
cpl = coupler(L = L, g = g, R = R, wg = wg_w, L_taper = L_taper)
cpl.plot()
cpl.show()
#%%
component_fiber_array = uc.add_fiber_array(component=cpl, fanout_length=100)
component_fiber_array.plot()
component_fiber_array.show()

Yellowshiba, 2024-09-04:
Follow up: Tried upgrad to version 8.8.6, issue not solved

joamatab, 2024-09-04:
can you `pip install ubcpdk --upgrade`? 

i just released a new version

joamatab, 2024-09-04:
this code worked for me. 

I also made you some improvements 

```python
import gdsfactory as gf
import ubcpdk.components as uc

# %%
@gf.cell
def DC_geo(L=30, g=0.07, R=5, wg=3, L_taper=20):
    c = (
        gf.Component()
    )  # Create a blank component (essentially an empty GDS cell with some special features)
    lda0 = 1.55
    """
    inf = L/2
    vertices_top = [
        (-inf, g/2 + wg),
        (inf, g/2 + wg),
        (inf, g/2),
        (-inf, g/2),
        
    ]
    ring_wg = c.add_polygon(vertices_top, layer=(1, 0))
    """
    vertices_bot = [
        (-L / 2 - L_taper - lda0, -g / 2 - wg / 2 + 0.25),
        (-L / 2 - L_taper, -wg / 2 - g / 2 + 0.25),
        (-L / 2, -g / 2),
        (L / 2, -g / 2),
        (L / 2 + L_taper, -g / 2 - wg / 2 + 0.25),
        (L / 2 + L_taper + lda0, -wg / 2 - g / 2 + 0.25),
        (L / 2 + L_taper + lda0, -wg / 2 - g / 2 - 0.25),
        (L / 2 + L_taper, -g / 2 - wg / 2 - 0.25),
        (L / 2, -g / 2 - wg),
        (-L / 2, -g / 2 - wg),
        (-L / 2 - L_taper, -g / 2 - wg / 2 - 0.25),
        (-L / 2 - L_taper - lda0, -g / 2 - wg / 2 - 0.25),
    ]
    c.add_polygon(vertices_bot, layer=(1, 0))
    c.add_port(
        name="o1",
        center=[-L / 2 - L_taper - lda0, -g / 2 - wg / 2],
        width=0.5,
        orientation=180,
        layer=(1, 0),
    )
    c.add_port(
        name="o2",
        center=[L / 2 + L_taper + lda0, -g / 2 - wg / 2],
        width=0.5,
        orientation=0,
        layer=(1, 0),
    )
    return c

# %%
@gf.cell
def coupler(L=30, g=0.07, R=5, wg=3, L_taper=20):
    c = gf.Component()
    lda0 = 1.55
    dc = c << DC_geo(L=L, g=g, R=R, wg=wg, L_taper=L_taper)
    bl = c << gf.components.bend_circular(
        angle=-90, cross_section="strip", allow_min_radius_violation=False
    )
    br = c << gf.components.bend_circular(
        angle=-90, cross_section="strip", allow_min_radius_violation=False
    )
    # gc_in = c << ubcpdk.components.gc_te1550()
    # gc_out = c << ubcpdk.components.gc_te1550()

    bl.connect("o2", dc.ports["o1"], allow_layer_mismatch=True)
    br.connect("o1", dc.ports["o2"], allow_layer_mismatch=True)
    # gc_in.connect("o1", bl.ports["o1"], allow_layer_mismatch=True)
    # gc_out.connect("o1", br.ports["o2"], allow_layer_mismatch=True)
    c.add_port(
        name="o1",
        center=[-L / 2 - L_taper - lda0 - R, -g / 2 - wg / 2 - R],
        width=0.5,
        orientation=180,
        layer=(1, 0),
    )
    c.add_port(
        name="o2",
        center=[L / 2 + L_taper + lda0 + R, -g / 2 - wg / 2 - R],
        width=0.5,
        orientation=0,
        layer=(1, 0),
    )
    return c

# %%
L = 260  # length pf the coupling region
g = 0.24
R = 5
wg_w = 3
L_taper = 40
cpl = coupler(L=L, g=g, R=R, wg=wg_w, L_taper=L_taper)

uc.add_fiber_array(component=cpl, fanout_length=100)
cpl.show()
```

closed 2024-09-06 by sebastian-goeldi