gdsfactory/gdsfactory #3190: ValueError: ref() is deprecated. Use add_ref() instead

tzarz, 2024-09-07:
Hi All,

I am new to gdsfactory and I've been trying to follow the instruction guide and I am receiving errors. I've struggled considerably installing the correct packages and making everything jive together, but now I think I am getting to the end. My current struggle is the error:

ValueError: ref() is deprecated. Use add_ref() instead

It seems as it calls out specifically the line get_sparameters_fdtdz. there is no 'ref' in the below code, so it must be coming from an issue upstream.

I have successfully installed fdtz, maybe there is a compatibility issue? I'm using gds 8.8.7, gplugins 1.1.2, the CUDA version of JAX, and tried to update all other packages to newest. Also using Python 3.11.0rc1

I am trying to follow guide here: https://gdsfactory.github.io/gplugins/notebooks/fdtdz_01.html
here is the code which I was trying to execute

import gdsfactory as gf
gf.config.print_version_plugins()
from gdsfactory.cross_section import rib
from gdsfactory.generic_tech import LAYER, LAYER_STACK
from gdsfactory.technology import LayerStack

from gplugins.fdtdz.get_sparameters_fdtdz import get_sparameters_fdtdz

length = 10

c = gf.Component()
waveguide_rib = c << gf.components.straight(length=length, cross_section=rib)
nitride_feature = c << gf.components.circle(radius=2, layer=LAYER.WGN)
nitride_feature.dx= 5
padding = c << gf.components.bbox(
    waveguide_rib, top=2, bottom=2, layer=LAYER.WAFER
)
c.add_ports(gf.components.straight(length=length).ports)

c.plot()

filtered_layer_stack = LayerStack(
    layers={
        k: LAYER_STACK.layers[k] for k in ["clad", "box", "core", "slab90", "nitride"]
    }
)
filtered_layer_stack

material_name_to_fdtdz = {
    "si": 3.45,
    "sio2": 1.44,
    "sin": 2.0,
}

out = get_sparameters_fdtdz(
    component=c,
    layer_stack=filtered_layer_stack,
    nm_per_pixel=20,
    extend_ports_length=0.0,
    zmin=-0.5,
    zz=96,
    tt=10000,
    wavelength=1.55,
    port_margin=1,
    material_name_to_fdtdz=material_name_to_fdtdz,
    default_index=1.44,
)

out

tzarz, 2024-09-07:
It seems similar to this case with meep:
https://github.com/gdsfactory/gdsfactory/discussions/2886

as of July 19th this is the progress, but fdtdz/jax is not listed:
https://github.com/gdsfactory/gplugins/issues/437

Could someone provide a current work around with some sort of compatibility list until the above work is completed? If I install directly gplugins, it installs the latest 8.8.3+ gdsfactory.

joamatab, 2024-10-11:
closing this issue as we are already tracking this here

https://github.com/gdsfactory/gplugins/issues/437

closed 2024-10-11 by joamatab