gdsfactory/gdsfactory #3192: Colab examples on docs site no longer run after gf requiring python 3.11

paulxshen, 2024-09-08:
Colab stuck on Python 3.10 but gdsfactory now requires 3.11 . Maybe roll back compat entry to 3.10?

joamatab, 2024-09-08:
Hi Paul,

This issue comes from Kfactory backend requiring python3.11 or larger

@sebastian-goeldi

sebastian-goeldi, 2024-09-10:
Could you try instead whether it would be feasible to install 3.11/12 in colab and see whether that works?

https://www.askpython.com/python/examples/update-python-google-colab

paulxshen, 2024-09-10:
Thanks Joaquin and Sebastian! I tried variations of it - need to restart session to apply updates but Colab then hangs :( could just wait/nudge on Colab to update to 3.11 lol

closed 2024-10-17 by joamatab

sebastian-goeldi, 2024-10-17:
Sure you want to close this? All the scientific packages start to drop 3.10 support in 2 months :-)