gdsfactory/gdsfactory #3197: Consistency of the commands.

GCShao, 2024-09-10:
**Describe the bug**
What's the bug?
The new version of gdsfactory doesnt support the earlier code command, which makes it super exhausting to maintain the projects that have been already done.

**Expected behavior**
What would you like to happen?
It is highly recommended that the new version should have a good compability with the old ones, because we dont want to spend too much efforts on how many commands have been changed over the past version.
Thank you!

joamatab, 2024-09-11:
Thank you for your recommendation!

We understand your concern, and we’ll make sure to communicate this feedback to the team. Ensuring that updates don't disrupt existing projects is a top priority, and maintaining stability for users like yourself is key.

If you prefer not to update, sticking with a stable version that supports your current projects might be the best option until a smoother transition plan is in place. Thanks again for your input!

closed 2024-09-11 by joamatab

GCShao, 2024-09-12:
> Thank you for your recommendation!
>
> We understand your concern, and we’ll make sure to communicate this feedback to the team. Ensuring that updates don't disrupt existing projects is a top priority, [...]

Thank you for your reply!As you suggest, I have sticked my project with gf v7 and found the corresponding manual. I appreciate the efforts of your team made constantly for improving this  great open source project!