gdsfactory/gdsfactory #3206: No module named 'kfactory.routing.generic'

Revenantx86, 2024-09-13:
Hello, I have recently reinstalled the klayout and sky130 pdk. And when I try to add tech file to klayout I get the following error: 

[image]

Can you explain how I can solve this issue ?  I have tried to switch to python3.11 but it doesnt seem to solve my issue. I cannot get klayout using python3.11

- [x ] I have reviewed the documentation but found no relevant solution.
- [x ] I have searched the existing issues and found no resolution.
- [x ] I am using the latest version of GDSFactory with Python 3.10, 3.11, or 3.12. Below is the output for the following code
- [x ] I acknowledge that, as an open-source project, the maintainers may not have the capacity to address this issue. I am willing to contribute by fixing it myself or hiring someone to resolve it (contact@gdsfactory.com). Otherwise, I understand that the issue may remain unresolved.

sebastian-goeldi, 2024-09-15:
Could you provide a little more information?

To clarify a bit, you shouldn't need to have the GUI KLayout withing the same environment as gdsfactory. Gdsfactory only needs the pypi package of klayout (i.e. the python standalone package), maybe this is not true for the sky130, but it should be. The interaction with klayout happens through a local only TCP connection which sends info to klayout about which file(s) to load.

To get a newer python environment than 3.10, I would suggest conda/mamba or if it should be something more lightweight [uv](https://github.com/astral-sh/uv) or pyenv are solid choices.

Revenantx86, 2024-09-17:
> Could you provide a little more information?
> 
> To clarify a bit, you shouldn't need to have the GUI KLayout within the same environment as gdsfactory. Gdsfactory only needs the pypi package of klayout (i.e. the python standalone package), maybe this is not true for the sky130, but it should be. The interaction with klayout GUI happens through a local only TCP connection which sends info to klayout about which file(s) to load.
> 
> To get a newer python environment than 3.10, I would suggest conda/mamba or if it should be something more lightweight [uv](https://github.com/astral-sh/uv) or pyenv are solid choices.

Hello, Thank you for your reply. I have found out that the issue was the sky130 cells were expecting the newer version of GDS-factory. However, the klayout build I have with ubuntu22 is set default to python3.10 I had to delete klayout and compile it with python3.11 or newer to be able to install newest version of GDS factory. After the build, all worked fine.

closed 2024-09-17 by Revenantx86