gdsfactory/gdsfactory #3220: get_route should error for width mismatch

joamatab, 2024-09-27:
This code should raise an error

```python
import gdsfactory as gf

c = gf.Component()
s1 = c << gf.c.straight(width=2)
s2 = c << gf.c.straight(width=2)

s2.dmove((20, 20))

gf.routing.route_single(
    component=c,
    port1=s1.ports["o1"],
    port2=s2.ports["o1"],
)

c.show()
```

and it doesn't 

[image]

joamatab, 2024-09-28:
this PR should fix it by adding automatic tapers

https://github.com/gdsfactory/gdsfactory/pull/3222

sebastian-goeldi, 2024-09-28:
Sure. I mean in the end I think it's fine however it's solved. connectivity check should find this as well

joamatab, 2024-09-28:
this is how it looks after PR

[image]

closed 2024-09-28 by joamatab