gdsfactory/gdsfactory #3225: import_gds ports loses layer infor

joamatab, 2024-09-28:
```python
    import gdsfactory as gf

    c = gf.components.mzi()
    c.pprint_ports()
    gdspath = c.write_gds()

    c = import_gds(gdspath)
    c.pprint_ports()
    c.show()
```
returns
```
┏━━━━━━┳━━━━━━━┳━━━━━━━━━━━━━┳━━━━━━━┳━━━━━━━━━━━━━━━━━━━━━━━━━━┳━━━━━━━━━━━┓
┃ name ┃ width ┃ orientation ┃ layer ┃ center                   ┃ port_type ┃
┡━━━━━━╇━━━━━━━╇━━━━━━━━━━━━━╇━━━━━━━╇━━━━━━━━━━━━━━━━━━━━━━━━━━╇━━━━━━━━━━━┩
│ o1   │ 0.5   │ 180.0       │ WG    │ (-10.0, 0.0)             │ optical   │
│ o2   │ 0.5   │ 0.0         │ WG    │ (81.10000000000001, 0.0) │ optical   │
└──────┴───────┴─────────────┴───────┴──────────────────────────┴───────────┘
┏━━━━━━┳━━━━━━━┳━━━━━━━━━━━━━┳━━━━━━━┳━━━━━━━━━━━━━━━━━━━━━━━━━━┳━━━━━━━━━━━┓
┃ name ┃ width ┃ orientation ┃ layer ┃ center                   ┃ port_type ┃
┡━━━━━━╇━━━━━━━╇━━━━━━━━━━━━━╇━━━━━━━╇━━━━━━━━━━━━━━━━━━━━━━━━━━╇━━━━━━━━━━━┩
│ o1   │ 0.5   │ 180.0       │ 0     │ (-10.0, 0.0)             │ optical   │
│ o2   │ 0.5   │ 0.0         │ 0     │ (81.10000000000001, 0.0) │ optical   │
└──────┴───────┴─────────────┴───────┴──────────────────────────┴───────────┘
```

@sebastian-goeldi

sebastian-goeldi, 2024-09-29:
use `port.layer_info` when printing the ports. This is an issue arising from the migration to support LayerInfos everywhere.

sebastian-goeldi, 2024-10-17:
is this still an issue?

joamatab, 2024-10-17:
yes,

i still get this

```
┏━━━━━━┳━━━━━━━┳━━━━━━━━━━━━━┳━━━━━━━┳━━━━━━━━━━━━━━━━━━━━━━━━━━┳━━━━━━━━━━━┓
┃ name ┃ width ┃ orientation ┃ layer ┃ center                   ┃ port_type ┃
┡━━━━━━╇━━━━━━━╇━━━━━━━━━━━━━╇━━━━━━━╇━━━━━━━━━━━━━━━━━━━━━━━━━━╇━━━━━━━━━━━┩
│ o1   │ 0.5   │ 180.0       │ 1     │ (-10.0, 0.0)             │ optical   │
│ o2   │ 0.5   │ 0.0         │ 1     │ (81.10000000000001, 0.0) │ optical   │
└──────┴───────┴─────────────┴───────┴──────────────────────────┴───────────┘
┏━━━━━━┳━━━━━━━┳━━━━━━━━━━━━━┳━━━━━━━┳━━━━━━━━━━━━━━━━━━━━━━━━━━┳━━━━━━━━━━━┓
┃ name ┃ width ┃ orientation ┃ layer ┃ center                   ┃ port_type ┃
┡━━━━━━╇━━━━━━━╇━━━━━━━━━━━━━╇━━━━━━━╇━━━━━━━━━━━━━━━━━━━━━━━━━━╇━━━━━━━━━━━┩
│ o1   │ 0.5   │ 180.0       │ 0     │ (-10.0, 0.0)             │ optical   │
│ o2   │ 0.5   │ 0.0         │ 0     │ (81.10000000000001, 0.0) │ optical   │
└──────┴───────┴─────────────┴───────┴──────────────────────────┴───────────┘
```

joamatab, 2024-10-17:
maybe we can copy the layer infos from the temp_kcl layout?

sebastian-goeldi, 2024-10-17:
maybe read my first message before confirming?

joamatab, 2024-10-17:
if i use layer_info i get this

```
┏━━━━━━┳━━━━━━━┳━━━━━━━━━━━━━┳━━━━━━━━━━┳━━━━━━━━━━━━━━━━━━━━━━━━━━┳━━━━━━━━━━━┓
┃ name ┃ width ┃ orientation ┃ layer    ┃ center                   ┃ port_type ┃
┡━━━━━━╇━━━━━━━╇━━━━━━━━━━━━━╇━━━━━━━━━━╇━━━━━━━━━━━━━━━━━━━━━━━━━━╇━━━━━━━━━━━┩
│ o1   │ 0.5   │ 180.0       │ WG (1/0) │ (-10.0, 0.0)             │ optical   │
│ o2   │ 0.5   │ 0.0         │ WG (1/0) │ (81.10000000000001, 0.0) │ optical   │
└──────┴───────┴─────────────┴──────────┴──────────────────────────┴───────────┘
┏━━━━━━┳━━━━━━━┳━━━━━━━━━━━━━┳━━━━━━━━━━━━━━━┳━━━━━━━━━━━━━━━━━━━━━━━━━━┳━━━━━━━━━━━┓
┃ name ┃ width ┃ orientation ┃ layer         ┃ center                   ┃ port_type ┃
┡━━━━━━╇━━━━━━━╇━━━━━━━━━━━━━╇━━━━━━━━━━━━━━━╇━━━━━━━━━━━━━━━━━━━━━━━━━━╇━━━━━━━━━━━┩
│ o1   │ 0.5   │ 180.0       │ WAFER (999/0) │ (-10.0, 0.0)             │ optical   │
│ o2   │ 0.5   │ 0.0         │ WAFER (999/0) │ (81.10000000000001, 0.0) │ optical   │
└──────┴───────┴─────────────┴───────────────┴──────────────────────────┴───────────┘
```

joamatab, 2024-10-17:
i think the problem is that we use temp_kcl layouts for loading GDS files, and those kclayouts  don't copy their info,

now we are copying over cross_sections

sebastian-goeldi, 2024-10-17:
I mean do what you want. But as far as I can see from the print above this fixes it...

joamatab, 2024-10-18:
the layer changes from WG to WAFER

sebastian-goeldi, 2024-10-18:
well, yes, because guess what layer index 0 is? :drum: 

```python
>>> import gdsfactory as gf
>>> gf.kcl.layers(0)
<LAYER.WAFER: 0>
```

joamatab, 2024-10-24:
fixed by https://github.com/gdsfactory/kfactory/pull/501

closed 2024-10-24 by joamatab