gdsfactory/gdsfactory #3227: avoid collisions

joamatab, 2024-09-28:
when using kfactory==0.20.5

[image]

avoiding colisions worked

however in latest kfactory==0.20.6 it doesn't

[image]

@sebastian-goeldi

joamatab, 2024-09-29:
i can confirm this issue is coming from autotaper

When i use `auto_taper=False` i get:

[image]

joamatab, 2024-10-01:
this feature will be in implemented in kfactory

closed 2024-10-01 by joamatab

sebastian-goeldi, 2024-10-01:
Yeah, but it's going to take time :D.

What you can do in the meantime is add
```python
box = gf.kdb.Box()
for taper in auto_tapers:
    box += auto_taper.ports["o1"].trans.disp.to_p()
    box += auto_taper.ports["o2"].trans.disp.to_p()
# call to route_bundle function with bboces = other_boxes + [box]
```