gdsfactory/gdsfactory #3228: Is there any detailed tutorials for user to build they own PDK？

luomeng007, 2024-09-29:
I have found someting on Website. If my understanding is correct, user should define theiry own LayerMap class.  However, what should I do after that? I mean is there any complete routine for this to learn. It was quite hard for me realise this.

Besides, I found plot() function can make use have a look on pcell in VS code Jupyter Notebook view.  Is this only valid for generic layers? Once I set layer like (54.0, 0.0), nothing was diplayed.

What should I do then?

joamatab, 2024-09-29:
You can take a look at some of the PDKs which are open source

https://github.com/gdsfactory/cspdk

You can modify this repository with the layers that you need

luomeng007, 2024-09-29:
Okay, Thank you for your replay, by the way, how about my second question? I mean when I set layer as (54.0, 0.0)， then when I use plot() function, nothing is shown in VS code Notebook view? Is this normal?

sebastian-goeldi, 2024-10-01:
Look at the si220 for example. That is essentially the pdk. Take a look at `__init__.py`. From there you should be able to follow what's going on. Start with the cells, the most important part. And maybe LAYER

doddgray, 2024-10-02:
What is the intended way to register cells, cross sections, etc. in a minimal custom PDK? Cell registration is mentioned as important but not demonstrated in the `PDK` section of the docs. It's not obvious to me how/where that is happening in GF maintained PDKs like `cspdk` and `si220`.

The `register_cells` PDK class method appears in the [YAML Place and AutoRoute](https://gdsfactory.github.io/gdsfactory/notebooks/10_yaml_component.html) docs page: ```gf.get_active_pdk().register_cells(pad_new=pad_new)```
but is otherwise undocumented. Is that not meant for public use when defining PDKs?

joamatab, 2024-10-02:
yes, that's a good point, maybe we should add a link to this PDK

https://github.com/gdsfactory/cspdk/blob/main/cspdk/si220/__init__.py

How does that sound?

luomeng007, 2024-10-08:
Thanks, guys, your suggestion is very helpful.

closed 2024-10-08 by luomeng007