gdsfactory/gdsfactory #3235: tap1.conect(xxx,xxx, overlap=xxx) have a bug when v7 migrate to v8

synesios, 2024-10-01:
**Describe the bug**
The statement tap1.connect('o1', ref.ports['o{}'.format((i + 1))], overlap=0.005) has encountered an issue. In version 7, the exported layout has no gaps at the connection points, but in version 8, the same statement generates a layout with gaps at the connection points. I tried increasing the overlap value, but it seems to have no effect.
[image]

joamatab, 2024-10-02:
Hi, 

we have a tutorial on how to define virtual cells
https://gdsfactory.github.io/gdsfactory/notebooks/12_grid_snapping.html

we don't use the overlap value anymore

synesios, 2024-10-02:
> Hi,
> 
> we have a tutorial on how to define virtual cells https://gdsfactory.github.io/gdsfactory/notebooks/12_grid_snapping.html
> 
> we don't use the overlap value anymore

I have already used the vcell, 
[image]
but it seems to have no effect,there still have a gap

synesios, 2024-10-02:
> Hi,
> 
> we have a tutorial on how to define virtual cells https://gdsfactory.github.io/gdsfactory/notebooks/12_grid_snapping.html
> 
> we don't use the overlap value anymore

`from lib import zjupdk
from functools import partial
import matplotlib.pyplot as plt
import gdsfactory as gf
import numpy as np
from numpy import *
from lib import zjucomponent
import warnings
warnings.filterwarnings("ignore")
from lib import zjucomponent as zj
from gdsfactory.cross_section import Section
# 预设参数
num1 = 32  # 阵列波导数量
num2 = 8  # 输出波导数量
m = 150  # 衍射级次
channel_spacing = 0.0008  # 通道间隔
lam = 1.55  # 中心波长
angel = 0  # 晶轴旋转角度

ra = 260  # 自由传播区半径
da = 3.4  # 阵列波导间距

Lt11 = 25  # 阵列波导入口四段长度
Lt12 = 5
Lt13 = 32
Lt14 = 25

width = 2.9  # 阵列波导入口taper宽度
Lt1 = Lt11+Lt12+Lt13+Lt14  # 阵列波导入口处taper长度加窄波导的总长度
width_s = 2  # 细波导宽度
Lt2 = 25  # 细波导出口taper长度
width_a = 1.5  # 阵列波导宽度
Lw = 140  # 细波导长度
width_in = 4.5  # 输入波导tapper宽度
Lt_in = 80  # 输入波导tapper长度
width_out = 4.5  # 输出波导tapper宽度
Lt_out = 80  # 输出波导tapper长度

r_2 = 100   # 弯曲波导r2
r_3 = 80.4   # 弯曲波导r3
Ls = 180  # S弯曲长度
s_0 = 2  # 路径s0长度
s_1 = 2  # 第一条路径s1长度
s_2 = 110  # 第一条路径s2长度

ng_ax = 2.19199  # 阵列波导x传群折射率  360刻180 宽度2 倾角60° 1550nm
ng_ay = 2.2964  # 阵列波导y传群折射率
ng_sx = 2.17376  # 平板波导x传群折射率  360刻180
ng_sy = 2.27159   # 平板波导y传群折射率

ne_ax = 1.81024  # 阵列波导x传有效折射率
ne_ay = 1.86955  # 阵列波导y传有效折射率
ne_sx = 1.83443  # 平板波导x传有效折射率
ne_sy = 1.89635  # 平板波导y传有效折射率

dr = 2*m*(ng_ax+ng_ay)*channel_spacing*ra/(ne_sx+ne_sy)/(ne_ax+ne_ay)/da  # 输出波导间距 通过计算获得
fsr = lam*(ne_ax + ne_ay)/m/(ng_ax + ng_ay)
# 定义自由传播区
theta_cal = ((((num1 + 1)/2-1)*da+width/2)/ra)*180/np.pi
# print(theta_cal)
theta_cal2 = (2*(((num2 + 2)/2-1)*dr+width/2)/ra)*180/np.pi
# print(theta_cal2)

theta1_min = -(theta_cal+10),
theta1_max = (theta_cal+10),
theta2_min = 180-(theta_cal2+10),
theta2_max= 180+(theta_cal2+10),
num= num1

c = gf.Component()
b = gf.Component()
point1 = zjucomponent.ellipse_arc(a=ra+0.005, b=ra+0.005, theta_max=theta1_max[0] , theta_min=theta1_min[0] )
point2 = zjucomponent.ellipse_arc(a=ra/2+0.5, b=ra/2+0.5, theta_max=theta2_max[0] , theta_min=theta2_min[0] )
translation_vector = np.array([ra/2, 0])
point3 = point2 + translation_vector
pts = np.vstack([point1, point3])

b.add_polygon(pts,  layer=(31, 0))
fsr1 = c.create_vinst(b)
for k in range(num1):
    c.add_port('o{}'.format((k + 1)),
            center=(ra * np.cos((((num1 + 1) / 2 - (k + 1)) * da) / ra),
                           ra * np.sin((((num1 + 1) / 2 - (k + 1)) * da) / ra)),
            orientation=((((num1 + 1) / 2 - (k + 1)) * da) / ra) * 180 / np.pi,
            width=width,
            layer=(31, 0))

w = gf.components.straight(length=10, width=2.9 )
w1 = c.create_vinst(w)
#w1.rotate(30)

w1.connect("o1",c.ports["o3"])
c.show()` 
 you can refind this problem through  my code.

sebastian-goeldi, 2024-10-02:
Hard to judge without a minimal reproducible example (currently traveling, so don't have access to my normal pc, but assuming from the imports, it's not possible with public accessible modules). But my guess is that c/b should also be ComponentAllAngle and mot Components, otherwise instances get snapped, latest on write

joamatab, 2024-11-15:
Hi, closing this issue as it's using an old version of GDSFactory and we don't have the code to reproduce it