gdsfactory/gdsfactory #3242: [Question] How to draw a straight waveguide PCell with polygon method?

luomeng007, 2024-10-08:
I have tried four methods to draw a straight waveguide pcell, like below:   
1. gf.kdb.DPolygon()  
2. add_polygon()  
3. gf.components.rectangle()
4. gf.path.straight()

All methods above can successfully draw a straight waveguide pcell. Howeverm if I export is to KLayout view. The figure like below:
[image]
See, only two points for constructing this straight waveguide. If I double left click on this pcell. The figure below appears, obviously, this is a box method to draw straight waveguide.
[image]
So, my question is, even I used polygon method to draw straight waveguide in GDSFactory, but when export straight waveguide into KLayout, the straight waveguie is build by using box method in KLayout.  But I want it is build by polygon method, like figure below:
[image]

Anyone can help me to deal with this? 

Besides, can we add a question label in issue, I only find enhancement and bug labels there!

sebastian-goeldi, 2024-10-08:
> Besides, can we add a question label in issue, I only find enhancement and bug labels there!

That's what discussions are for ;)

The reason you see a box is that this is an optimization from klayout. A box is internally just a more straight forward representation of a rectangle polygon (a box you can describe by two points instead of using 4 when using a polygon).

You can force klayout however to not do this optimization (not sure why you would ever want to do that, but here goes):

## pure dbu case (i.e. work in database units, normally this is `1 dbu == 1nm`, i.e. `dbu == 0.001`)

```python
import kfactory as kf

pts = [
    kf.kdb.Point(0, 0),
    kf.kdb.Point(5, 0),
    kf.kdb.Point(10, 0),
    kf.kdb.Point(15, 0),
    kf.kdb.Point(20, 0),
    kf.kdb.Point(25, 0),
    kf.kdb.Point(25, 10),
    kf.kdb.Point(20, 10),
    kf.kdb.Point(15, 10),
    kf.kdb.Point(10, 10),
    kf.kdb.Point(5, 10),
    kf.kdb.Point(0, 10),
]

d1 = kf.kdb.Polygon(pts)
print(f"{d1=}")

d2 = kf.kdb.Polygon(
    pts,
    raw=True,
)
print(f"{d2=}")

c = kf.KCell()
p = c.shapes(kf.kdb.LayerInfo(1, 0)).insert(d1)
print(p)
p = c.shapes(kf.kdb.LayerInfo(1, 0)).insert(d2)
print(p)
```

```
d1=(0,0;0,10;25,10;25,0)
d2=(0,0;0,10;5,10;10,10;15,10;20,10;25,10;25,0;20,0;15,0;10,0;5,0)
polygon (0,0;0,10;25,10;25,0)
polygon (0,0;0,10;5,10;10,10;15,10;20,10;25,10;25,0;20,0;15,0;10,0;5,0)
```

As you can see the optimization happens already  when creating the polygon, if we pass `raw=True`, it will tell KLayout to not optimize it.

## um based version

`raw=True` doesn't have an effect on um based DPolygons (I assume because it would be inefficient and inaccurate in many cases). But we can coerce the DPolygon to a Polygon first which doesn't optimize the polygon and keeps the points.

```python
import kfactory as kf

pts = [
    kf.kdb.DPoint(0, 0),
    kf.kdb.DPoint(5, 0),
    kf.kdb.DPoint(10, 0),
    kf.kdb.DPoint(15, 0),
    kf.kdb.DPoint(20, 0),
    kf.kdb.DPoint(25, 0),
    kf.kdb.DPoint(25, 10),
    kf.kdb.DPoint(20, 10),
    kf.kdb.DPoint(15, 10),
    kf.kdb.DPoint(10, 10),
    kf.kdb.DPoint(5, 10),
    kf.kdb.DPoint(0, 10),
]

d1 = kf.kdb.DPolygon(pts)
print(f"{d1=}")

c = kf.KCell()
p = c.shapes(kf.kdb.LayerInfo(1, 0)).insert(d1)
print(p)
p = c.shapes(kf.kdb.LayerInfo(1, 0)).insert(d1.to_itype(c.kcl.dbu))
print(p)
```

```
d1=(0,0;0,10;5,10;10,10;15,10;20,10;25,10;25,0;20,0;15,0;10,0;5,0)
polygon (0,0;0,10000;25000,10000;25000,0)
polygon (0,0;0,10000;5000,10000;10000,10000;15000,10000;20000,10000;25000,10000;25000,0;20000,0;15000,0;10000,0;5000,0)
```

sebastian-goeldi, 2024-10-08:
You can however force the optimization of DPolygon with `dpolygon.compress(True)` if you want to do that on DPolgyons