gdsfactory/gdsfactory #3263: Meep Plugin for gds factory does not work properly

sohil1234, 2024-10-11:
**Describe the bug**  
When running the provided documentation code for generating s-parameters using Meep, I encounter a runtime error:  
`RuntimeError: meep: Source width > cell width in y direction!`

**To Reproduce**  
The error can be reproduced with the following code:

```python
import gdsfactory as gf
import matplotlib.pyplot as plt
import numpy as np
from autograd import tensor_jacobian_product
from gdsfactory.generic_tech import get_generic_pdk
from meep import MaterialGrid, Medium, Vector3, Volume
from meep.adjoint import (
    DesignRegion,
    get_conic_radius_from_eta_e,
)

import gplugins
import gplugins.gmeep as gm

gf.config.rich_output()
PDK = get_generic_pdk()
PDK.activate()

c = gf.components.taper(length=2.0, width1=0.25, width2=0.25)
c.plot()
sp = gm.write_sparameters_meep(c, run=True)
```

The complete error traceback is as follows:
``RuntimeError: meep: Source width > cell width in y direction!``

**Environment (please complete the following information):**

- [x] I have reviewed the documentation but found no relevant solution.
- [x] I have searched the existing issues and found no resolution.
- [x] I am using the latest version of GDSFactory with Python 3.10, 3.11, or 3.12. Below is the output for the following code
- [x] I acknowledge that, as an open-source project, the maintainers may not have the capacity to address this issue. I am willing to contribute by fixing it myself or hiring someone to resolve it (contact@gdsfactory.com). Otherwise, I understand that the issue may remain unresolved.

```python
import sys
print(sys.version)
print(sys.executable)

import gdsfactory as gf
gf.config.print_version_plugins()
```

The output from the above code is:
| Package    | Version                                         | Path                                               |
|------------|-------------------------------------------------|----------------------------------------------------|
| python     | 3.11.10 (main, Oct  3 2024, 07:29:13) [GCC 11.2.0] | /home/phoenix/miniconda3/envs/photonics/bin/python |
| gdsfactory | 8.12.0                                          | /home/phoenix/miniconda3/envs/photonics/lib/python3.11/site-packages |
| gplugins   | 1.1.2                                           | /home/phoenix/miniconda3/envs/photonics/lib/python3.11/site-packages |
| ray        | not installed                                   |                                                    |
| femwell    | 0.1.11                                          | /home/phoenix/miniconda3/envs/photonics/lib/python3.11/site-packages |
| devsim     | 2.8.4                                           | /home/phoenix/miniconda3/envs/photonics/lib/python3.11/site-packages |
| tidy3d     | 2.7.4                                           | /home/phoenix/miniconda3/envs/photonics/lib/python3.11/site-packages |
| meep       | 1.29.0                                          | /home/phoenix/miniconda3/envs/photonics/lib/python3.11/site-packages |
| meow       | 0.11.2                                          | /home/phoenix/miniconda3/envs/photonics/lib/python3.11/site-packages |
| lumapi     | not installed                                   |                                                    |
| sax        | 0.13.3                                          | /home/phoenix/miniconda3/envs/photonics/lib/python3.11/site-packages |

simbilod, 2024-10-11:
Have you tried increasing the padding or reducing the port size (not advisable)? If I recall this error shows up if your simulation domain is smaller than the MEEP port that gets defined

joamatab, 2024-10-11:
Hi Simon,

the meep plugin does not work at all since the migration from 7 to 8,

if you have some time it would be great to fix this

https://github.com/gdsfactory/gplugins/issues/437

sohil1234, 2024-10-11:
So except for tidy3d does any other fdtd solver plugin works ?

joamatab, 2024-10-12:
I have only tested the tidy3d, which is the one i've been using

I started fixing the Lumerical one as well but it may take some time
https://github.com/gdsfactory/gplugins/pull/486

PRs are always welcome

sohil1234, 2024-10-12:
Which version of Gplugins was supported by gds factory 7

joamatab, 2024-11-15:
Closing this for now as this should be addressed in https://github.com/gdsfactory/gplugins/issues

closed 2024-11-15 by joamatab