gdsfactory/gdsfactory #3274: Klive can't open .gds when using c.show()

GiomB, 2024-10-14:
Hello, I am using gdsfactory 8.13.3 with Klayout 0.29.7, Python 3.12 and Ubuntu 24.04.
While using klayout with klive working, I use c.show() and I receive this error :

`ERROR Unable to open file: /tmp//1977097362.gds (errno=2) in MainWindow.load_layout`

Then Klive crashes.

I don't find the good documentation on the c.show() function and would like to know if it's possible to store the temporary .gds file elsewhere ?
I don't even find the 1977097362.gds file in my /tmp/ folder.

Thank you for the help :)

joamatab, 2024-10-14:
yes, you can do `c.write_gds('hello.gds')`

GiomB, 2024-10-15:
Thanks for you reply !
The function c.write_gds is just writing the gds file, klive doesn't automatically update klayout.
I can't do `c.show('demo.gds)` , klive still tries to find the file in /tmt/ which doesn't seem to exist ...

tvt173, 2024-10-15:
hi @GiomB , i think what you want to do is this:

```python
import gdsfactory as gf

c = gf.c.rectangle()
filename = "hello.gds"
c.write_gds(filename)
gf.show(filename)
```

sebastian-goeldi, 2024-10-15:
@GiomB could you share the env? I suspenct this might be an issue with jupyter. I.e. are you running this in a jupyter and also conda?

GiomB, 2024-10-15:
Hello @tvt173 , indeed with the function gf.show, i can specify the path and it works nicely ! Thank you for this workaround :)
But still I am wondering why the c.show function is not working as intended. On windows, the gds file is indeed created in appdata/local/temp. Maybe something wrong with my tmp folder on my ubuntu ?

@sebastian-goeldi I am using python with virtualenvs and then VSCode for the notebooks
Here is my env:

```
3.12.6 (main, Nov 10 2011, 15:00:00) [GCC 14.2.0]
/home/giom/.virtualenvs/Laser/bin/python
```

<pre style="white-space:pre;overflow-x:auto;line-height:normal;font-family:Menlo,'DejaVu Sans Mono',consolas,'Courier New',monospace"><span style="font-style: italic">                                                      Modules                                                      </span>
┏━━━━━━━━━━━━┳━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┳━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┓
┃<span style="font-weight: bold">    Package </span>┃<span style="font-weight: bold"> version                                         </span>┃<span style="font-weight: bold">                                             Path </span>┃
┡━━━━━━━━━━━━╇━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━╇━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┩
│<span style="color: #008080; text-decoration-color: #008080">     python </span>│<span style="color: #800080; text-decoration-color: #800080"> 3.12.6 (main, Nov 10 2011, 15:00:00) [GCC       </span>│<span style="color: #008000; text-decoration-color: #008000">         /home/giom/.virtualenvs/Laser/bin/python </span>│
│<span style="color: #008080; text-decoration-color: #008080">            </span>│<span style="color: #800080; text-decoration-color: #800080"> 14.2.0]                                         </span>│<span style="color: #008000; text-decoration-color: #008000">                                                  </span>│
│<span style="color: #008080; text-decoration-color: #008080"> gdsfactory </span>│<span style="color: #800080; text-decoration-color: #800080"> 8.13.3                                          </span>│<span style="color: #008000; text-decoration-color: #008000"> /home/giom/.virtualenvs/Laser/lib/python3.12/si… </span>│
│<span style="color: #008080; text-decoration-color: #008080">   gplugins </span>│<span style="color: #800080; text-decoration-color: #800080"> not installed                                   </span>│<span style="color: #008000; text-decoration-color: #008000">                                                  </span>│
│<span style="color: #008080; text-decoration-color: #008080">        ray </span>│<span style="color: #800080; text-decoration-color: #800080"> not installed                                   </span>│<span style="color: #008000; text-decoration-color: #008000">                                                  </span>│
│<span style="color: #008080; text-decoration-color: #008080">    femwell </span>│<span style="color: #800080; text-decoration-color: #800080"> not installed                                   </span>│<span style="color: #008000; text-decoration-color: #008000">                                                  </span>│
│<span style="color: #008080; text-decoration-color: #008080">     devsim </span>│<span style="color: #800080; text-decoration-color: #800080"> not installed                                   </span>│<span style="color: #008000; text-decoration-color: #008000">                                                  </span>│
│<span style="color: #008080; text-decoration-color: #008080">     tidy3d </span>│<span style="color: #800080; text-decoration-color: #800080"> not installed                                   </span>│<span style="color: #008000; text-decoration-color: #008000">                                                  </span>│
│<span style="color: #008080; text-decoration-color: #008080">       meep </span>│<span style="color: #800080; text-decoration-color: #800080"> not installed                                   </span>│<span style="color: #008000; text-decoration-color: #008000">                                                  </span>│
│<span style="color: #008080; text-decoration-color: #008080">       meow </span>│<span style="color: #800080; text-decoration-color: #800080"> not installed                                   </span>│<span style="color: #008000; text-decoration-color: #008000">                                                  </span>│
│<span style="color: #008080; text-decoration-color: #008080">     lumapi </span>│<span style="color: #800080; text-decoration-color: #800080"> not installed                                   </span>│<span style="color: #008000; text-decoration-color: #008000">                                                  </span>│
│<span style="color: #008080; text-decoration-color: #008080">        sax </span>│<span style="color: #800080; text-decoration-color: #800080"> not installed                                   </span>│<span style="color: #008000; text-decoration-color: #008000">                                                  </span>│
└────────────┴─────────────────────────────────────────────────┴──────────────────────────────────────────────────┘
</pre>

Thank you all for the help

sebastian-goeldi, 2024-10-17:
Hmm. It's interesting that it picks such a curious name for the gds. Normally it defaults to the cellname. I am not sure why in your case it will try to write a numeric name. Anyway, I am not sure why you can't write to `/tmp`. Are you on a univeristy network or similar? I remember at ETHZ we had a weird temp (though I have to admit that was in the very early days of my linux endeavors). But nonetheless, I can offer two potential solutions.

1. I can implement `KFACTORY_BUILD_DIR` to allow you to set the folder used for saving to wherever you want (either through env variable directly or it is also loaded from an env file)
2. Create a git repository, this will make kfactory default to build into `{git_repo_root}/build/gds`

luomeng007, 2024-10-25:
I also encounted a case like this. I use VS code, when I use `cell.show()`. The VS code tell me 

> could not connect to klive serve

And my solution is to open KLayout in advance, and then use `cell.show()`, it can draw corresponding layout inside KLayout. And I can not find klive extension in VS code. So is there a method to let VS code open KLayout automatically and draw layout?

sebastian-goeldi, 2024-10-25:
Theoretically there is. But a) I don't like if programs just open ither programs without asking, b) this could spawn tons of klayout processes in cases where for exampls klive is broken. Maybe a muddle groubd would be to ask the user whether klayout should be started?

luomeng007, 2024-10-28:
I totally agree. 

And maybe another solution is to make eye-catching tips or instructions about this to let user know exactly what they should do to get result when they use `cell.show()`  function. :)