gdsfactory/gdsfactory #3296: Importing gdsfactory breaks JSON schema validation

edelmanjm, 2024-10-18:
**Describe the bug**

Given an otherwise totally functional JSON schema validation function, importing gdsfactory in the main results in an unexpected JSON schema validation error:

```
jsonschema.exceptions.ValidationError: ('VDD1P2', 'VDD2P2', 'VDD5P0', 'GND') is not of type 'array'

Failed validating 'type' in schema['properties']['voltages']:
    {'type': 'array'}

On instance['voltages']:
    ('VDD1P2', 'VDD2P2', 'VDD5P0', 'GND')
```

**To Reproduce**

- Download these files
- Remove the .txt extensions, if applicable
- Run demo.py and observe the failure. To see it pass, simply remove or comment out `import gdsfactory` in demo.py:6.

[base.schema.json](https://github.com/user-attachments/files/17427976/base.schema.json)
[demo.schema.json](https://github.com/user-attachments/files/17427977/demo.schema.json)
[demo.py](https://github.com/user-attachments/files/17427984/demo.py.txt)
[demo.yaml](https://github.com/user-attachments/files/17427985/demo.yaml.txt)

**Expected behavior**
- The schema validation still works after importing gdsfactory

**Suggested fix**
- Unsure, unfortunately

**Environment (please complete the following information):**

- [x] I have reviewed the documentation but found no relevant solution.
- [x] I have searched the existing issues and found no resolution.
- [x] I acknowledge that, as an open-source project, the maintainers may not have the capacity to address this issue. I am willing to contribute by fixing it myself or hiring someone to resolve it (contact@gdsfactory.com). Otherwise, I understand that the issue may remain unresolved.
- [x] I am using the latest version of GDSFactory with Python 3.11, or 3.12. Below is the output for the following code

```python
import sys
print(sys.version)
print(sys.executable)

import gdsfactory as gf
gf.config.print_version_plugins()
```

```
                                    Modules                                     
┏━━━━━━━━━━━━┳━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┳━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┓
┃    Package ┃ version                        ┃                           Path ┃
┡━━━━━━━━━━━━╇━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━╇━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┩
│     python │ 3.12.4 (main, Jun  6 2024,     │ /Users/edelmanjm/Projects/top… │
│            │ 18:26:44) [Clang 15.0.0        │                                │
│            │ (clang-1500.3.9.4)]            │                                │
│ gdsfactory │ 8.9.1                          │ /Users/edelmanjm/Projects/top… │
│   gplugins │ not installed                  │                                │
│        ray │ not installed                  │                                │
│    femwell │ not installed                  │                                │
│     devsim │ not installed                  │                                │
│     tidy3d │ not installed                  │                                │
│       meep │ not installed                  │                                │
│       meow │ not installed                  │                                │
│     lumapi │ not installed                  │                                │
│        sax │ not installed                  │                                │
└────────────┴────────────────────────────────┴────────────────────────────────┘
```

sebastian-goeldi, 2024-10-18:
I think this is related or a duplicate of https://github.com/gdsfactory/gdsfactory/issues/3188

edelmanjm, 2024-10-18:
Ah, you're right, I didn't see that one since I was searching for "JSON" instead of "YAML". I think that's a duplicate.

Do we happen to have a timeline on a fix?

joamatab, 2024-10-18:
Thank you @edelmanjm  and Orion for pointing this out

See PR here with the fix
https://github.com/gdsfactory/gdsfactory/pull/3298

closed 2024-10-18 by joamatab