gdsfactory/gdsfactory #3310: Cell decorator changes order of gf.components.text() when passing a dictionary

tgustafson-ntnu, 2024-10-23:
**Describe the bug**
If I pass a `dict` into a function with a cell decorator, the text order changes. If I pass a `str` there is not a problem. If I remove the `@gf.cell` decorator, there is not a problem. This does not seem intentional.

**To Reproduce**
```python
import gdsfactory as gf

@gf.cell
def dic_box(value:dict)-> gf.Component:
    text1 = str(value).replace(', ','\n',).replace("'",'').replace('{','').replace('}','')+'\n'
    info = gf.components.text(
        text=text1,
        size=1,justify="left", layer=(1, 0), position= (0,0))
    return info

@gf.cell
def str_box(value:str)-> gf.Component:
    info = gf.components.text(
        text=value, 
        size=1,justify="left", layer=(1, 0), position= (0,0))
    return info

text_params = {'one':1, 'two':2, 'three':3,
               'four':4, 'five':5, 'six':6,
               'seven':7, 'eight':8, 'nine':9}
text1 = str(text_params).replace(', ','\n',).replace("'",'').replace('{','').replace('}','')+'\n'

dic_box(text_params).plot()
str_box(text1).plot()

```
**Expected behavior**
An ordered text list. 

**Suggested fix**
Not sure. When `gf.components.text()` is called, there is a cell decorator, so the cell decorator appears to duplicate, but I am unsure why that would/should change the order of the text.  

**Environment (please complete the following information):**

- [x] I have reviewed the documentation, discussions and issues and found no relevant solution.
- [x] I understand that this is an open-source project, and maintainers may not have the resources to address every issue. I am prepared to contribute by fixing the issue myself or hiring someone to do so, if needed, and accept that the issue may not be resolved otherwise.
- [x] I am using the latest version of GDSFactory, with Python 3.10, 3.11, or 3.12.

Please provide the output for the following code:

```python
python │ 3.12.6 (tags/v3.12.6:a4a2d2b, Sep  6 2024,  20:11:23) [MSC v.1940 64 bit (AMD64)]        
gdsfactory │ 8.15.0                                         
gplugins, ray, femwell, devsim, tidy3d, meep, meow, lumapi, sax │ not installed 
```

sebastian-goeldi, 2024-10-23:
This seems to be a sideeffect of using frozenset as a vessel for making dictionaries hashable for the decorator. At the moment I don't have a better idea, but maybe there is something that can be done, no promises though. If need to keep the order of the dictionary, please use something hashable like a tuple of tuples or similar. I never noticed this because generally I never rely on dictionaries being ordered objects (they are also not really promised to be that in python, they just are since version 3 something afaik)

sebastian-goeldi, 2024-10-23:
On the other hand this means the following will not create two separate cells:

```python
import gdsfactory as gf

@gf.cell
def my_cell_with_a_dict(a: dict[str, str]) -> gf.Component:
    c = gf.Component()
    return c

a1 = {"hello": "a", "bye": "b"}
a2 = {"bye": "b", "hello": "a"}

print(my_cell_with_a_dict(a1).cell_index() == my_cell_with_a_dict(a2).cell_index())
```

```
True
```