gdsfactory/gdsfactory #3313: cannot install gdsfactory 8.15.0 in Klayout through Package Manager

cheng1214, 2024-10-26:
Hi!

When I tried to install gdsfactory 8.15.0 in Klayout, it shows an error message 'error fetching spec file for package gdsfactory.....' (see picture below)
[image]

Can anyone help me?
Thanking you!

sebastian-goeldi, 2024-10-26:
Klayout's salt mine is atm often the target of DoS attacks. Try again later, this is especially a problem from Asia if I understood right see https://klayout.de/forum/discussion/2602/sami-package-management

cheng1214, 2024-10-26:
Thanks for the reply. I just tried again and it still doesn't work. I however can install other package nornally in Klayout.

Can I install this mannually?
Thanks!

closed 2024-10-26 by joamatab

joamatab, 2024-10-26:
yes, you can

after `pip install gdsfactory --upgrade` you can open a terminal and run

` gf install-klayout-genericpdk  `

make sure you are using gdsfactory 8.15.1 or later

cheng1214, 2024-10-26:
Thanks! I followed the instruction. But I use 8.15.0 ...  How to check if it works? I see a new toolbar in my Klayout (as below) appearing but no drop-down list available from that

[image]

cheng1214, 2024-10-26:
Hi,  I a new starter for gdsfactory. As I said above, how to check if that has been installed properly? 

I just ran some example codes, which do not give any error except a warning that " _Could not connect to klive server_". Also NO .gds is updated on my Klayout as well.  

Thanking you!

cheng1214, 2024-10-26:
I have updated to 8.15.1, but still no gds is updated, and the warning becomes "_klive didn't send data, closing_"