gdsfactory/gdsfactory #3317: klive not work as expected

cheng1214, 2024-10-26:
Hi!

When I run some example codes with c.show ( ), no gds appears on my Klayout, and a warning "_kfactory.kcell.shows:8843: klive didn't send data, closing_" appears
 wondered how I can solve this problem? The gdsfactory version is 8.15.1. I

Thanking you!

joamatab, 2024-10-26:
we can't help you with more information

did you give connection permissions to klayout when you opened it?
are you using mac os, windows or linux?
what version of klayout are you using?

can you see klive menu the UI?

[image]

sebastian-goeldi, 2024-10-26:
This can also be helpful in case it doesn't work:

https://gdsfactory.github.io/klive/troubleshooting/

cheng1214, 2024-10-27:
_Did you grant connection permissions to KLayout when you opened it?_

I don't know how to do that. Could you please give me some instructions?

_Are you using macOS, Windows, or Linux?_

I am using Windows 10

 _Which version of KLayout are you using?_

The version is 0.27.5

_Can you see the "KLive" menu in the UI?_

Yes, it shows like below. Before runing the 'c.show ( )', KLive is in "live state", but aftet that, it becomes to RECV state.
[image]

cheng1214, 2024-10-27:
When I run "klive" in Macro Development in Klayout, it shows message:
klive 0.3.3 stopped
klive 0.3.3 is running

sebastian-goeldi, 2024-10-27:
Use a newer klayout verions. On 0.27 it certainly will not run. That's also why you were getting the error, klive and meta ports need at least 0.29.

Your klive is crashing while receiving the gds or klayout cannot read it. Please install a klayout > 0.29, ideally the latest

cheng1214, 2024-10-28:
Hi,

Thanks for the Info., and it works after upgrading Klayout to 0.29.8. 

I however didn't find `Show/Hide Port` menu that I saw in the tutorials in my Klayoyt. I wondered how to install that?

sebastian-goeldi, 2024-10-28:
It's from the second klayout package in the gdsfactory package (that one is more or less  just a package to automatically install both)

Please try again to install through the gdsfactory package. It should install the other package.

joamatab, 2024-10-29:
I think I found the issue

See fix here
https://github.com/gdsfactory/metainfo-ports/pull/4

joamatab, 2024-10-29:
If you update to the latest gdsfactory it should work now

https://sami.klayout.org/index

Go to Klayout packages --> update

closed 2024-10-29 by joamatab

cheng1214, 2024-10-29:
I have updated to gdsfactory 8.16.0 (which should be the latest one) in Klayout . But I still don't see Show/Hide Port menu (as below)
[image]

sebastian-goeldi, 2024-10-29:
I noticed gdsfactory was still pointing to the old now inexistent metainfo-port package. Can you uninstall the metainfo package and reinstall through the gdsfactory package please?