gdsfactory/gdsfactory #3327: easy improvements

joamatab, 2024-10-29:
- [ ] improve schematic visualization (graphviz) instead of networkx
- [ ] improve routing https://github.com/gdsfactory/gdsfactory/issues/57
- [ ] bring back[ a-star router ](https://github.com/gdsfactory/gdsfactory7/blob/main/gdsfactory/routing/get_route_astar.py)

QuentinWach, 2024-11-12:
Could you specify what you mean with "improve integration with electronics tools"? I can imagine a whole list of things to do and I am curious. @joamatab

juanferv, 2025-06-19:
Sorry I posted in other issue
Hi @joamatab

Greetings from the Valencian photonics ecosystem! We truly appreciate the work behind GDSFactory — it’s an incredible platform, and we’d like to contribute as much as we can that might be useful for PIC designers.

My colleague Daniel has developed a prototype repo of an automated tool to assist in the design of integrated optical splitters (MMIs) with low-loss arbitrary splitting: 
https://github.com/daniBerAs/FLAMA 

We're not software experts, so the code architecture might have room for improvement, but the goal is to provide a parametric design framework that serves as a solid starting point for more advanced optimization techniques.

If you find it interesting, we’d be happy if you can help us to evolve this idea to the Next level!

Best regards

joamatab, 2025-06-23:
Thank you Juan!  looks great!