gdsfactory/gdsfactory #3331: `c.plot( )` or `scene.show()` do not work in .py files

cheng1214, 2024-11-01:
Hi, 

I wondred if either `c.plot( )` or `scene.show()` only work in  .ipynb, but not in .py files?

In my case, when using `c.plot( )` in a .py file, no figure is popped up, while using `scene.show()` in a .py file,  an error "ModuleNotFoundError: No module named 'pyglet'" was raised. `c.show()` however works normally.

All of them works however well in .ipynb files...

sebastian-goeldi, 2024-11-02:
Because ipynb/ipython do some magic behind the scenes. But I believe you can force it to show a figure in normal python with `return_fig=True` as an argumen.

ubc downgrades because here it has a specific version requirement for gdsfactory  https://github.com/gdsfactory/ubc/blob/main/pyproject.toml#L18

cheng1214, 2024-11-03:
Thanks the reply. 
But `c.plot(return_fig=True)` still doesn't work and no figure popped up

sebastian-goeldi, 2024-11-03:
Yes it returns a matplotlib figure in that case, the showing must be handled by you as far as I can tell. Not an expert though, I never use this

Neuromystix, 2024-11-05:
I had the same issue on Spyder (Anaconda installation), which uses Ipython. If i run the full code with `F5`, containing some `c.plot(return_fig=True)`, the plot is ignored by Spyder. If then I run just the `c.plot(return_fig=True)` line with `F9`, it is plotted ok to _inline_. I assumed something behind the Ipython backend or whatever.  

I ended up using just `c.show()`.

closed 2024-11-10 by joamatab